riscv-software-src/homebrew-riscv #3: riscv-tools homebrew build fails on OS X El Cap.

riferguson, 2015-12-23:
The failure I'm seeing is similar to the previous issue with OS X Yosemite; a cascading build failure terminating in ":configure: error: C preprocessor "/lib/cpp" fails sanity check" during the build of build/build-gcc-newlib/riscv64-unknown-elf/libgcc. I've attached the contents of the Logs/Homebrew/riscv-gcc/03.make file for context.

I'm running OS X 10.11.2, with Xcode 7.2.

Thanks.

[03.make.txt](https://github.com/riscv/homebrew-riscv/files/71241/03.make.txt)

sbeamer, 2015-12-29:
Your error appears similar to what Travis is also reporting. On my system running 10.11.2 everything works after downgrading the packages used by gcc (last week's commit). I'll hopefully be able to reproduce and fix this in the next few weeks.

sbeamer, 2016-01-05:
Please retry (after running brew update) and let me know if I fixed it. The bottles are now El Capitan, so it should use those (much faster install), but it should also be able to build from source too.

sbeamer, 2016-07-10:
Please comment again to re-open this issue.

closed 2016-07-10 by sbeamer