riscv-software-src/homebrew-riscv #4: Empty Installation?

markercraig, 2016-02-18:
I follow the instructions exactly, and end up receiving this error: 

==> Installing riscv-tools from riscv/riscv
==> Downloading https://github.com/riscv/riscv-tools/archive/homebrew.tar.gz
Already downloaded: /Library/Caches/Homebrew/riscv-tools-0.1.tar.gz
Error: Empty installation

Any thoughts?

alansuphd, 2016-02-19:
I have exactly the same problem. "brew update" twice then fix all warnings reported by "brew doctor" and the problem persists. Any feedback/solutions are highly appreciated.

sbeamer, 2016-02-19:
Thank you for the report. This was caused by a commit to Homebrew (https://github.com/Homebrew/homebrew/commit/915ac06db21acc5c9595b2223cc39b9672f34c4a) which Travis didn't detect because it is running an older version of Homebrew. We use riscv-tools as an alias to install the needed repos and contain a trivial hello world test, but it is an "empty installation" as now blocked by Homebrew.

commit ca50562a (2016-02-19): fix #4 by copying over 1 file
closed 2016-02-19 by sbeamer