riscv-software-src/homebrew-riscv #6: This brew doesn't install riscv-tests

hauhsu, 2016-07-11:
Hi, 
I am trying Rocket Chip Generator. 
The README says mac users can install riscv-tools using homebrew. 
But the homebrew version doesn't install riscv-tests so that I can't run tests 
with commands like `make -j4 run-asm-tests`.
Should this be an issue or I just have to compile the version included in submodule of rocket-chip repo?

Thanks.

sbeamer, 2016-07-11:
I would compile the version in the rocket-chip repo for two reasons:

- The Makefiles within rocket-chip will look for riscv-tests in that location
- The submodule pointers within the repo should hopefully point to the appropriate versions

Let me know how it goes.

hauhsu, 2016-08-17:
Yeah I should compile the riscv-tools in rocket-chip repo.
Thanks a lot!

closed 2016-08-17 by hauhsu