riscv-software-src/homebrew-riscv #7: Build fails on macOS with clang: ../fesvr/tsi.cc:34:7: error: use of undeclared identifier 'abort'

josharian, 2016-11-28:
```bash
$ brew install riscv-tools
==> Installing riscv-tools from riscv/riscv
==> Installing dependencies for riscv/riscv/riscv-tools: riscv-fesvr, riscv-isa-sim, gmp, mpfr, gawk, gnu-sed, libmpc, isl, riscv-gcc, riscv-pk
==> Installing riscv/riscv/riscv-tools dependency: riscv-fesvr
==> Cloning https://github.com/riscv/riscv-fesvr.git
Cloning into '/Users/josh/Library/Caches/Homebrew/riscv-fesvr--git'...
remote: Counting objects: 53, done.
remote: Compressing objects: 100% (52/52), done.
remote: Total 53 (delta 1), reused 22 (delta 1), pack-reused 0
Unpacking objects: 100% (53/53), done.
==> Checking out branch master
==> ../configure --prefix=/usr/local/Cellar/riscv-fesvr/fesvr
==> make prefix=/usr/local
Last 15 lines from /Users/josh/Library/Logs/Homebrew/riscv-fesvr/02.make:
clang++ -fPIC -MMD -MP  -DTARGET_ARCH="" -Wall -O2 -std=c++11 -DPREFIX=\"/usr/local\" -I. -I../fesvr -c ../fesvr/rfb.cc
clang++ -fPIC -MMD -MP  -DTARGET_ARCH="" -Wall -O2 -std=c++11 -DPREFIX=\"/usr/local\" -I. -I../fesvr -c ../fesvr/context.cc
clang++ -fPIC -MMD -MP  -DTARGET_ARCH="" -Wall -O2 -std=c++11 -DPREFIX=\"/usr/local\" -I. -I../fesvr -c ../fesvr/htif_pthread.cc
clang++ -fPIC -MMD -MP  -DTARGET_ARCH="" -Wall -O2 -std=c++11 -DPREFIX=\"/usr/local\" -I. -I../fesvr -c ../fesvr/htif_hexwriter.cc
clang++ -fPIC -MMD -MP  -DTARGET_ARCH="" -Wall -O2 -std=c++11 -DPREFIX=\"/usr/local\" -I. -I../fesvr -c ../fesvr/dummy.cc
clang++ -fPIC -MMD -MP  -DTARGET_ARCH="" -Wall -O2 -std=c++11 -DPREFIX=\"/usr/local\" -I. -I../fesvr -c ../fesvr/option_parser.cc
clang++ -fPIC -MMD -MP  -DTARGET_ARCH="" -Wall -O2 -std=c++11 -DPREFIX=\"/usr/local\" -I. -I../fesvr -c ../fesvr/term.cc
clang++ -fPIC -MMD -MP  -DTARGET_ARCH="" -Wall -O2 -std=c++11 -DPREFIX=\"/usr/local\" -I. -I../fesvr -c ../fesvr/tsi.cc
clang++ -fPIC -MMD -MP  -DTARGET_ARCH="" -Wall -O2 -std=c++11 -DPREFIX=\"/usr/local\" -I. -I../fesvr -c ../fesvr/elf2hex.cc
../fesvr/tsi.cc:34:7: error: use of undeclared identifier 'abort'
      abort();
      ^
1 error generated.
make: *** [tsi.o] Error 1
make: *** Waiting for unfinished jobs....
```

```bash
$ clang --version
Apple LLVM version 8.0.0 (clang-800.0.42.1)
Target: x86_64-apple-darwin16.1.0
Thread model: posix
InstalledDir: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin
```

Probably related: https://github.com/riscv/riscv-fesvr/issues/18

Thanks!

aswaterman, 2016-11-29:
> Probably related: riscv/riscv-fesvr#18

Can you confirm this is resolved by
https://github.com/riscv/riscv-fesvr/pull/20 ?

josharian, 2016-11-29:
Confirmed.

josharian, 2016-11-29:
Fixed by #20.

closed 2016-11-29 by josharian