riscv-software-src/homebrew-riscv #8: riscv-gcc build fails with clang and no gcc: configure: error: C preprocessor "/lib/cpp" fails sanity check

josharian, 2016-11-29:
```bash
$ brew install riscv-tools
==> Installing riscv-tools from riscv/riscv
==> Installing dependencies for riscv/riscv/riscv-tools: riscv-gcc, riscv-pk
==> Installing riscv/riscv/riscv-tools dependency: riscv-gcc
==> Cloning https://github.com/riscv/riscv-gnu-toolchain.git
Updating /Users/josh/Library/Caches/Homebrew/riscv-gcc--git
==> Checking out branch master
==> ../configure --prefix=/usr/local/Cellar/riscv-gcc/gnu
==> make
Last 15 lines from /Users/josh/Library/Logs/Homebrew/riscv-gcc/02.make:
libtool: link: (cd ".libs" && rm -f "libcc1.so" && ln -s "libcc1.0.so" "libcc1.so")
libtool: link: sed -e 's,^,_,' < /private/tmp/riscv-gcc-20161128-36691-176ds08/build/src/newlib-gcc/libcc1/libcc1plugin.sym > .libs/libcc1plugin-symbols.expsym
libtool: link: /usr/bin/clang++ -Wl,-undefined -Wl,dynamic_lookup -o .libs/libcc1plugin.0.so -bundle  .libs/plugin.o .libs/callbacks.o .libs/connection.o .libs/marshall.o   -L/usr/local/lib  -Wl,-no_pie -F/usr/local/Frameworks -Wl,-headerpad_max_install_names -isysroot /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.12.sdk ../libiberty/pic/libiberty.a   -Wl,-exported_symbols_list,.libs/libcc1plugin-symbols.expsym
libtool: link: ( cd ".libs" && rm -f "libcc1.la" && ln -s "../libcc1.la" "libcc1.la" )
libtool: link: dsymutil .libs/libcc1plugin.0.so || :
libtool: link: (cd ".libs" && rm -f "libcc1plugin.so" && ln -s "libcc1plugin.0.so" "libcc1plugin.so")
libtool: link: ( cd ".libs" && rm -f "libcc1plugin.la" && ln -s "../libcc1plugin.la" "libcc1plugin.la" )
unsupported
checking how to run the C preprocessor... /lib/cpp
configure: error: in `/private/tmp/riscv-gcc-20161128-36691-176ds08/build/build-gcc-newlib/riscv64-unknown-elf/libgcc':
configure: error: C preprocessor "/lib/cpp" fails sanity check
See `config.log' for more details.
make[2]: *** [configure-target-libgcc] Error 1
make[1]: *** [all] Error 2
make: *** [stamps/build-gcc-newlib] Error 2
```

```bash
$ clang --version
Apple LLVM version 8.0.0 (clang-800.0.42.1)
Target: x86_64-apple-darwin16.1.0
Thread model: posix
InstalledDir: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin
$ cpp --version
Apple LLVM version 8.0.0 (clang-800.0.42.1)
Target: x86_64-apple-darwin16.1.0
Thread model: posix
InstalledDir: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin
```

Sorry I don't have the cycles to hunt this down further myself.

sbeamer, 2016-11-29:
I believe the recent commits to riscv-fesvr fix this. Can you try it again?

sbeamer, 2016-11-29:
I wasn't able to reproduce this. Running `brew install riscv-gcc` completes successfully (after quite a bit of time). Additionally, manually cloning riscv-gnu-toolchain and building there also works. I just upgraded to macOS Sierra, and I'll hopefully be able to push new pre-built bottles in a few days.

What happens when you try to manually build it?

josharian, 2016-11-30:
I just tried again; still no dice.

If by manually build it, you mean follow along the original brew recipe steps, I get the same result. I haven't tried any other means of building it; did you have a particular method in mind?

Superficially it appears that, for whatever reason, the build system isn't detecting my c++ compiler.

> I just upgraded to macOS Sierra

It's also possible that it's not just Sierra, but also the fact that I'm doing this on a totally clean OS install, so no gcc or old clang versions or libraries lying around anywhere.

sbeamer, 2016-11-30:
By manually, I mean cloning  riscv-gnu-toolchain directly and following its build instructions.

I agree with your hypothesis that it might be an issue of detecting your compiler. I would test it out on other code just to be sure. After upgrading, I had to run xcrun a few times to get all of the dev tools working.

If nothing else, hopefully the bottles will help.

josharian, 2016-11-30:
The brew install made it through several other compilations before dying on this one; I'm pretty sure my compiler is generally working and detectable. Will try the direct clone+build and report back.

josharian, 2016-11-30:
I get a different failure with the direct clone+build. I had to manually point to my brew-installed gmp, mpfr, and libmpc.

```bash
$ git clone --recurse-submodules git@github.com:riscv/riscv-gnu-toolchain.git
$ cd riscv-gnu-toolchain
$ export CPPFLAGS="-I/usr/local/opt/gmp/include -I/usr/local/opt/mpfr/include -I/usr/local/opt/libmpc/include"
$ export LDFLAGS="-L/usr/local/opt/gmp/lib -L/usr/local/opt/mpfr/lib -L/usr/local/opt/libmpc/lib"
$ ./configure --prefix=/Users/josh/blarg
$ make linux
//
// *lots* of output elided, available on request
//
riscv64-unknown-linux-gnu-gcc -m64 divmod_1.c -c -std=gnu11 -fgnu89-inline  -O2 -Wall -Wundef -Wwrite-strings -fmerge-all-constants -frounding-math -g -Wstrict-prototypes -Wold-style-definition     -ftls-model=initial-exec    -I/usr/local/opt/gmp/include -I/usr/local/opt/mpfr/include -I/usr/local/opt/libmpc/include  -I../include -I/Users/josh/blarg/riscv-gnu-toolchain/build-glibc-linux64/stdlib  -I/Users/josh/blarg/riscv-gnu-toolchain/build-glibc-linux64  -I../sysdeps/unix/sysv/linux/riscv/rv64  -I../sysdeps/unix/sysv/linux/riscv  -I../sysdeps/riscv/nptl  -I../sysdeps/unix/sysv/linux/generic  -I../sysdeps/unix/sysv/linux/wordsize-64  -I../sysdeps/unix/sysv/linux/include -I../sysdeps/unix/sysv/linux  -I../sysdeps/nptl  -I../sysdeps/pthread  -I../sysdeps/gnu  -I../sysdeps/unix/inet  -I../sysdeps/unix/sysv  -I../sysdeps/unix  -I../sysdeps/posix  -I../sysdeps/riscv/rv64  -I../sysdeps/riscv  -I../sysdeps/init_array  -I../sysdeps/ieee754/flt-32  -I../sysdeps/ieee754/dbl-64  -I../sysdeps/ieee754/ldbl-128  -I../sysdeps/riscv/soft-fp  -I../sysdeps/wordsize-64  -I../sysdeps/riscv/fpu  -I../sysdeps/ieee754  -I../sysdeps/generic  -I.. -I../libio -I.   -D_LIBC_REENTRANT -include /Users/josh/blarg/riscv-gnu-toolchain/build-glibc-linux64/libc-modules.h -DMODULE_NAME=libc -include ../include/libc-symbols.h       -o /Users/josh/blarg/riscv-gnu-toolchain/build-glibc-linux64/stdlib/divmod_1.o -MD -MP -MF /Users/josh/blarg/riscv-gnu-toolchain/build-glibc-linux64/stdlib/divmod_1.o.dt -MT /Users/josh/blarg/riscv-gnu-toolchain/build-glibc-linux64/stdlib/divmod_1.o
In file included from divmod_1.c:27:0:
/usr/local/opt/gmp/include/gmp.h:326:33: error: expected declaration specifiers or '...' before '(' token
 #define __GMP_CAST(type, expr)  ((type) (expr))
                                 ^
/usr/local/opt/gmp/include/gmp.h:1499:21: note: in expansion of macro '__GMP_CAST'
   mpn_divrem_1 (qp, __GMP_CAST (mp_size_t, 0), np, nsize, dlimb)
                     ^~~~~~~~~~
divmod_1.c:43:1: note: in expansion of macro 'mpn_divmod_1'
 mpn_divmod_1 (mp_ptr quot_ptr,
 ^~~~~~~~~~~~
make[3]: *** [/Users/josh/blarg/riscv-gnu-toolchain/build-glibc-linux64/stdlib/divmod_1.o] Error 1
make[2]: *** [stdlib/subdir_lib] Error 2
make[1]: *** [all] Error 2
make: *** [stamps/build-glibc-linux64] Error 2
```

sbeamer, 2016-12-02:
For building gcc, I would recommend building the newlib version (just make instead of make linux). That is what the homebrew-riscv is building. Building the glibc (linux) version of riscv-gcc takes longer, and is awkward to do on OS X due to the need for a case-sensitive file system (workaround is making a case sensitive disk image for the build). If you run into the case issue, the build errors will mention files with suffix .os.

You shouldn't have to tell configure where to find gmp, mpfr, and libmpc, and that might be a hint about the issue you are experiencing. On my system, the dev tools version of clang (default, aliased as g++), has no problem finding the homebrew installed libraries.

If nothing else, I would try installing from brew again. I uploaded bottles for Sierra, so it should use those and not even try to build.

josharian, 2016-12-02:
The bottles work. I'll close this, since all I really care about is being able to use the tools. Thanks for your help.

closed 2016-12-02 by josharian

sbeamer, 2016-12-02:
I'm sorry I couldn't fully resolve your issue. There have been a few issues like this where it has been difficult to remotely debug an issue with brew. Usually, getting it to work on Travis is a good sign, but currently that is failing due to timeouts.