riscv-software-src/homebrew-riscv #9: riscv-tests cannot be compiled with bottles

brettcannon, 2017-01-04:
When trying to compile riscv/riscv-tests from a git checkout, I'm getting the following error using the tools from the tap:
```
/Applications/Xcode.app/Contents/Developer/usr/bin/make -C isa -f /Users/brettcannon/Repositories/harmony/riscv-tests/isa/Makefile src_dir=/Users/brettcannon/Repositories/harmony/riscv-tests/isa XLEN=64
riscv64-unknown-elf-gcc -march=rv32g -mabi=ilp32 -static -mcmodel=medany -fvisibility=hidden -nostdlib -nostartfiles -I/Users/brettcannon/Repositories/harmony/riscv-tests/isa/../env/p -I/Users/brettcannon/Repositories/harmony/riscv-tests/isa/macros/scalar -T/Users/brettcannon/Repositories/harmony/riscv-tests/isa/../env/p/link.ld rv32ui/simple.S -o rv32ui-p-simple
riscv64-unknown-elf-gcc: error: unrecognized command line option '-mabi=ilp32'
make[1]: *** [rv32ui-p-simple] Error 1
make: *** [isa] Error 2
```

Are the bottles perhaps out of date?

sbeamer, 2017-01-04:
Yes, the bottles are almost certainly out of date. The most recent flurry of changes to riscv-gcc have been the ABI and arch command line arguments. Can you try rebuilding from source and see if that fixes the problem?
`brew reinstall --build-from-source riscv-gcc`

For riscv-tests, you may find you need to also install md5sha1sum.

brettcannon, 2017-01-04:
Now I get another error message:
```
riscv64-unknown-elf-gcc -march=rv32g -mabi=ilp32 -static -mcmodel=medany -fvisibility=hidden -nostdlib -nostartfiles -I/Users/brettcannon/Repositories/harmony/riscv-tests/isa/../env/p -I/Users/brettcannon/Repositories/harmony/riscv-tests/isa/macros/scalar -T/Users/brettcannon/Repositories/harmony/riscv-tests/isa/../env/p/link.ld rv32ui/sll.S -o rv32ui-p-sll
rv32ui/../rv64ui/sll.S: Assembler messages:
rv32ui/../rv64ui/sll.S:34: Error: value of 71144870016 too large for field of 4 bytes at 568
rv32ui/../rv64ui/sll.S:34: Error: value of 71144870016 too large for field of 4 bytes at 572
rv32ui/../rv64ui/sll.S:35: Error: value of 9106543362048 too large for field of 4 bytes at 600
rv32ui/../rv64ui/sll.S:36: Error: value of 1193612851550355456 too large for field of 4 bytes at 628
rv32ui/../rv64ui/sll.S:42: Error: value of 71144870016 too large for field of 4 bytes at 720
rv32ui/../rv64ui/sll.S:42: Error: value of 71144870016 too large for field of 4 bytes at 724
rv32ui/../rv64ui/sll.S:43: Error: value of 9106543362048 too large for field of 4 bytes at 752
make[1]: *** [rv32ui-p-sll] Error 1
make: *** [isa] Error 2
```

So at least part of the issue is due to the bottles being out-of-date.

sbeamer, 2017-01-04:
This may be an issue with riscv-tests. Briefly skimming the commit log shows some activity around masking constants. Perhaps the version of riscv-tests you have is inconsistent, or somehow the constant masking isn't working on macOS.

brettcannon, 2017-01-04:
OK, I'll report the issue over there. Thanks!

closed 2017-01-04 by sbeamer