riscv-software-src/homebrew-riscv #10: Error: riscv/riscv/riscv-tools: failed

gitfoxi, 2017-07-06:
```
$ brew test riscv-tools
Testing riscv/riscv/riscv-tools
==> Using the sandbox
==> riscv64-unknown-elf-gcc /tmp/riscv-tools-test-20170706-38532-1gd40un/hello.c
==> spike pk /tmp/riscv-tools-test-20170706-38532-1gd40un/a.out
Error: riscv/riscv/riscv-tools: failed
undefined method `assertions' for #<Formulary::FormulaNamespace9da3d5cffade8db99d366e0ec1a5e269::RiscvTools:0x007fa28936dbb0>
/Users/mfox/.gem/ruby/2.0.0/gems/minitest-5.5.1/lib/minitest/assertions.rb:126:in `assert'
/System/Library/Frameworks/Ruby.framework/Versions/2.0/usr/lib/ruby/2.0.0/test/unit/assertions.rb:38:in `assert'
/System/Library/Frameworks/Ruby.framework/Versions/2.0/usr/lib/ruby/2.0.0/test/unit/assertions.rb:187:in `assert_equal'
/usr/local/Homebrew/Library/Homebrew/formula_assertions.rb:10:in `shell_output'
/usr/local/Homebrew/Library/Taps/riscv/homebrew-riscv/riscv-tools.rb:28:in `block in <class:RiscvTools>'
/usr/local/Homebrew/Library/Homebrew/formula.rb:1639:in `block (2 levels) in run_test'
/usr/local/Homebrew/Library/Homebrew/formula.rb:832:in `with_logging'
/usr/local/Homebrew/Library/Homebrew/formula.rb:1638:in `block in run_test'
/usr/local/Homebrew/Library/Homebrew/extend/fileutils.rb:14:in `block in mktemp'
/usr/local/Homebrew/Library/Homebrew/extend/fileutils.rb:74:in `block in run'
/usr/local/Homebrew/Library/Homebrew/extend/fileutils.rb:74:in `chdir'
/usr/local/Homebrew/Library/Homebrew/extend/fileutils.rb:74:in `run'
/usr/local/Homebrew/Library/Homebrew/extend/fileutils.rb:13:in `mktemp'
/usr/local/Homebrew/Library/Homebrew/formula.rb:1632:in `run_test'
/usr/local/Homebrew/Library/Homebrew/test.rb:28:in `block in <main>'
/System/Library/Frameworks/Ruby.framework/Versions/2.0/usr/lib/ruby/2.0.0/timeout.rb:66:in `timeout'
/usr/local/Homebrew/Library/Homebrew/test.rb:27:in `<main>'
```

First try. Not after installing using README instructions. Seems more like its failing to run the test than failing a real test?

sbeamer, 2017-07-06:
I'm not able to reproduce it. Anything you can do to help debug would be greatly appreciated.

Based on the error message, it seems possible homebrew changed internally such that the formula (riscv-tools.rb) is using an outdated internal homebrew API for assertion checking. Unfortunately, with a fresh install, I do not get that error.

gitfoxi, 2017-07-06:
My fault or homebrew's. Looking at the backtrace, it's using a Gem from my
~/.gem and that can't be right. I just blew away that path and now it's
fine (hope I don't need gems for anything).

Thanks, and sorry for the spurious report.

closed 2017-07-06 by gitfoxi

sbeamer, 2017-07-07:
Glad it worked out :)