riscv-software-src/homebrew-riscv #11: error trying to exec 'cc1': execvp: No such file or directory

Vasu-p, 2017-07-28:
The installation completed without any errors. But when I am trying to compile a Hello World program it gives this message:

`riscv64-unknown-elf-gcc: error trying to exec 'cc1': execvp: No such file or directory`

But when I try to do the same from the /usr/local/opt/riscv-gcc/bin directory it works fine.

sbeamer, 2017-08-08:
I'm not able to reproduce this error. Searching online for it suggests a number of possible causes. At first glance, perhaps a permission issue is the most problem? Did you install it with sudo?

sbeamer, 2017-08-30:
Just to clarify, you shouldn't need to use sudo to install riscv-tools. Homebrew for the last few years hasn't needed sudo.

Since there has been no response in 3 weeks, I'm closing this issue. If your problem persists, please comment again to re-open it.

madScript01, 2017-09-09:
I'm getting that error while compiling a kernel and still couldn't find a solution :(

sbeamer, 2017-09-25:
@madScript01 @Vasu-p Either of you get a chance to double check on this?

madScript01, 2017-10-04:
Fixed this,  using linaro GCC toolchain and after typing apt-get install aarch64-linux-gnu

sbeamer, 2017-10-04:
@madScript01 Glad to hear your issues were resolved. Did you experience the issue originally with homebrew on macOS, or were you always using linux (linaro)?

closed 2018-01-03 by sbeamer