riscv-software-src/homebrew-riscv #12: riscv64-unknown-elf ar: ...No such file or directory

mtariq86, 2017-10-09:
I followed the instructions and when I installed the riscv-tools, it gave me error at this point:

==> Checking out branch master
==> ../configure --prefix=/usr/local/Cellar/riscv-gcc/gnu
==> make
Last 15 lines from /Users/TariqAzmy/Library/Logs/Homebrew/riscv-gcc/02.make:
riscv64-unknown-elf-ar: ../signal/lib.a: No such file or directory
riscv64-unknown-elf-ar: ../time/lib.a: No such file or directory
riscv64-unknown-elf-ar: ../locale/lib.a: No such file or directory
riscv64-unknown-elf-ar: ../reent/lib.a: No such file or directory
riscv64-unknown-elf-ar: ../errno/lib.a: No such file or directory
riscv64-unknown-elf-ar: ../misc/lib.a: No such file or directory
riscv64-unknown-elf-ar: ../machine/lib.a: No such file or directory
riscv64-unknown-elf-ar: *.o: No such file or directory
make[6]: *** [libc.a] Error 1
make[5]: *** [all-recursive] Error 1
make[4]: *** [all-recursive] Error 1
make[3]: *** [all] Error 2
make[2]: *** [all-target-newlib] Error 2
make[1]: *** [all] Error 2
make: *** [stamps/build-newlib] Error 2

Any idea what might cause this? Thanks

sbeamer, 2017-10-12:
Thank you for reporting this. Can you go through the logs /Users/TariqAzmy/Library/Logs/Homebrew/riscv-gcc/02.make) and attempt to find the first error? It looks like gcc is failing to build libc.

mtariq86, 2017-10-13:
Thanks for your reply.

**I skimmed through the log and here is what I found:**

In file included from /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.13.sdk/usr/include/sys/_types.h:32:0,
                 from /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.13.sdk/usr/include/sys/param.h:79,
                 from /private/tmp/riscv-gcc-20171009-12775-yuxl72/build/../riscv-newlib/newlib/libc/search/hash_bigkey.c:37:
/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.13.sdk/usr/include/sys/cdefs.h:761:2: error: #error Unsupported architecture
 #error Unsupported architecture

**If you are looking for the error when it does the make, here is what I found:**

make[6]: *** [lib_a-argz_count.o] Error 1
make[6]: *** Waiting for unfinished jobs....
In file included from /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.13.sdk/usr/include/stdlib.h:63:0,
                 from /private/tmp/riscv-gcc-20171009-12775-yuxl72/build/../riscv-newlib/newlib/libc/argz/argz_add_sep.c:9:

sbeamer, 2017-10-16:
I still don't know the issue, but it looks like it might be related to High Sierra. Unfortunately, due to a release deadline, I won't be able to upgrade my system to High Sierra until December. At which point, I could hopefully debug this issue. In the mean time, I might try compiling riscv-gcc directly.

sbeamer, 2018-01-03:
I believe I have fixed this issue with the latest set of commits to support High Sierra. @mtariq86, please let me know if it works.

sbeamer, 2018-03-02:
Please re-open this if your issue persists.

closed 2018-03-02 by sbeamer