riscv-software-src/homebrew-riscv #13: Install fails on 10.11.6 (El Capitan): undeclared identifier 'NAN'

aprnath, 2017-10-13:
I followed instruction on El Capitan (10.11.6), and get the following message:

% brew install riscv-tools
==> Installing riscv-tools from riscv/riscv
==> Installing dependencies for riscv/riscv/riscv-tools: riscv-isa-sim, riscv-gcc, riscv-pk
==> Installing riscv/riscv/riscv-tools dependency: riscv-isa-sim
==> Cloning https://github.com/riscv/riscv-isa-sim.git
Updating /Users/ajay/Library/Caches/Homebrew/riscv-isa-sim--git
==> Checking out branch master
==> ../configure --prefix=/usr/local/Cellar/riscv-isa-sim/isa
==> make install
Last 15 lines from /Users/ajay/Library/Logs/Homebrew/riscv-isa-sim/02.make:
    /usr/bin/install -c -m 444 $file /usr/local/Cellar/riscv-isa-sim/isa/include/spike; \
  done
clang++ -fPIC -MMD -MP  -Wall -Wno-unused -g -O2 -std=c++11 -DPREFIX=\"/usr/local/Cellar/riscv-isa-sim/isa\" -I. -I.. -I../riscv -I../dummy_rocc -I../softfloat -I../spike_main -c ../riscv/processor.cc
clang++ -fPIC -MMD -MP  -Wall -Wno-unused -g -O2 -std=c++11 -DPREFIX=\"/usr/local/Cellar/riscv-isa-sim/isa\" -I. -I.. -I../riscv -I../dummy_rocc -I../softfloat -I../spike_main -c ../riscv/execute.cc
clang++ -fPIC -MMD -MP  -Wall -Wno-unused -g -O2 -std=c++11 -DPREFIX=\"/usr/local/Cellar/riscv-isa-sim/isa\" -I. -I.. -I../riscv -I../dummy_rocc -I../softfloat -I../spike_main -c ../riscv/sim.cc
clang++ -fPIC -MMD -MP  -Wall -Wno-unused -g -O2 -std=c++11 -DPREFIX=\"/usr/local/Cellar/riscv-isa-sim/isa\" -I. -I.. -I../riscv -I../dummy_rocc -I../softfloat -I../spike_main -c ../riscv/interactive.cc
**../riscv/interactive.cc:250:59: error: use of undeclared identifier 'NAN'
  fprintf(stderr, "%g\n", isBoxedF32(f.r) ? (double)f.s : NAN);**
                                                          ^
**../riscv/interactive.cc:257:51: error: use of undeclared identifier 'NAN'
  fprintf(stderr, "%g\n", isBoxedF64(f.r) ? f.d : NAN);**
                                                  ^
2 errors generated.
make: *** [interactive.o] Error 1
make: *** Waiting for unfinished jobs....

Any ideas?

sbeamer, 2017-10-15:
I would try it again. I think this was a problem upstream for riscv-isa-sim, and they just fixed [it](https://github.com/riscv/riscv-isa-sim/commit/a91d9f7d89abd6cda6fafb7b5e7cacf4b3590c29).

aprnath, 2017-10-16:
Thanks sbeamer. Indeed, it builds with the latest updates.

closed 2017-10-16 by aprnath