riscv-software-src/homebrew-riscv #14: /bin/sh: /private/tmp/riscv-gcc-20171019-29012-12cjijs/build/../riscv-gcc/configure: No such file or directory

JackieHanC, 2017-10-19:
when I run 'brew install riscv-tools', it shows that after 'make'.
"Last 15 lines from /Users/jackie/Library/Logs/Homebrew/riscv-gcc/02.make:
--enable-languages=c,c++ \
		--with-system-zlib \
		--with-newlib \
		--disable-libmudflap \
		--disable-libssp \
		--disable-libquadmath \
		--disable-libgomp \
		--disable-nls \
		--enable-checking=yes \
		--disable-multilib \
		--with-abi=lp64d \
		--with-arch=rv64imafdc \
		CFLAGS_FOR_TARGET="-Os  -mcmodel=medlow"
/bin/sh: /private/tmp/riscv-gcc-20171019-29012-12cjijs/build/../riscv-gcc/configure: No such file or directory
make: *** [stamps/build-gcc-newlib-stage1] Error 127"

I don't know what's wrong, please help

JackieHanC, 2017-10-20:
It may be a problem with macOS High Sierra.

sbeamer, 2017-10-23:
Could you try building riscv-gcc directly (on High Sierra)? It is worth confirming that is possible before fixing Homebrew.

sbeamer, 2018-01-03:
I believe I have fixed this issue with the latest set of commits to support High Sierra. @JackieHanC, please let me know if it works.

JackieHanC, 2018-01-04:
Yeah, it works, thanks ! @sbeamer

closed 2018-01-04 by sbeamer