riscv-software-src/homebrew-riscv #15: Can't pass brew test

hauhsu, 2018-02-10:
Hi,
After installing riscv-tools with homebrew, I can't pass the test:
 
```
$ brew test riscv-tools
Testing riscv/riscv/riscv-tools
==> riscv64-unknown-elf-gcc /tmp/riscv-tools-test-20180210-7034-1hg2eep/hello.c
Last 15 lines from /Users/hhsu/Library/Logs/Homebrew/riscv-tools/test.01.riscv64-unknown-elf-gcc:
2018-02-10 22:43:58 +0800

riscv64-unknown-elf-gcc
/tmp/riscv-tools-test-20180210-7034-1hg2eep/hello.c

dyld: Library not loaded: /usr/local/opt/mpfr/lib/libmpfr.4.dylib
  Referenced from: /usr/local/Cellar/riscv-gcc/gnu/libexec/gcc/riscv64-unknown-elf/6.1.0/cc1
  Reason: image not found
riscv64-unknown-elf-gcc: internal compiler error: Abort trap: 6 (program cc1)
Please submit a full bug report,
with preprocessed source if appropriate.
See <http://gcc.gnu.org/bugs.html> for instructions.
Error: riscv/riscv/riscv-tools: failed
Failed executing: riscv64-unknown-elf-gcc /tmp/riscv-tools-test-20180210-7034-1hg2eep/hello.c
/usr/local/Homebrew/Library/Homebrew/formula.rb:1792:in `block in system'
/usr/local/Homebrew/Library/Homebrew/formula.rb:1730:in `open'
/usr/local/Homebrew/Library/Homebrew/formula.rb:1730:in `system'
/usr/local/Homebrew/Library/Taps/riscv/homebrew-riscv/riscv-tools.rb:28:in `block in <class:RiscvTools>'
/usr/local/Homebrew/Library/Homebrew/formula.rb:1636:in `block (3 levels) in run_test'
/usr/local/Homebrew/Library/Homebrew/utils.rb:551:in `with_env'
/usr/local/Homebrew/Library/Homebrew/formula.rb:1635:in `block (2 levels) in run_test'
/usr/local/Homebrew/Library/Homebrew/formula.rb:835:in `with_logging'
/usr/local/Homebrew/Library/Homebrew/formula.rb:1634:in `block in run_test'
/usr/local/Homebrew/Library/Homebrew/extend/fileutils.rb:14:in `block in mktemp'
/usr/local/Homebrew/Library/Homebrew/extend/fileutils.rb:74:in `block in run'
/usr/local/Homebrew/Library/Homebrew/extend/fileutils.rb:74:in `chdir'
/usr/local/Homebrew/Library/Homebrew/extend/fileutils.rb:74:in `run'
/usr/local/Homebrew/Library/Homebrew/extend/fileutils.rb:13:in `mktemp'
/usr/local/Homebrew/Library/Homebrew/formula.rb:1628:in `run_test'
/usr/local/Homebrew/Library/Homebrew/test.rb:28:in `block in <main>'
/System/Library/Frameworks/Ruby.framework/Versions/2.3/usr/lib/ruby/2.3.0/timeout.rb:91:in `block in timeout'
/System/Library/Frameworks/Ruby.framework/Versions/2.3/usr/lib/ruby/2.3.0/timeout.rb:33:in `block in catch'
/System/Library/Frameworks/Ruby.framework/Versions/2.3/usr/lib/ruby/2.3.0/timeout.rb:33:in `catch'
/System/Library/Frameworks/Ruby.framework/Versions/2.3/usr/lib/ruby/2.3.0/timeout.rb:33:in `catch'
/System/Library/Frameworks/Ruby.framework/Versions/2.3/usr/lib/ruby/2.3.0/timeout.rb:106:in `timeout'
/usr/local/Homebrew/Library/Homebrew/test.rb:27:in `<main>'
```

Is that because the version of mpfr library doesn't match?
```
$ ls /usr/local/opt/mpfr/lib/
libmpfr.6.dylib libmpfr.a       libmpfr.dylib   pkgconfig
```
I am running in macOS 10.13.3
Thanks

hauhsu, 2018-02-18:
It works! Thank a lot :))

closed 2018-02-19 by sbeamer

VijayrajS, 2018-03-08:
How long does the command take?
When I used
`brew reinstall --build-from-source riscv-gcc`
It is stuck in
`==> Reinstalling riscv/riscv/riscv-gnu-toolchain 
==> Cloning https://github.com/riscv/riscv-gnu-toolchain.git
Updating /Users/vijayrajs/Library/Caches/Homebrew/riscv-gnu-toolchain--git
==> Checking out branch master` 
indefinitely

sbeamer, 2018-03-08:
It can take a very long time depending on your internet connection speed and whether you are being throttled by Github. It is cloning the gcc git repo, which has quite a few files. Ideally it should take less than 30 minutes, but to be safe I would leave it running overnight.

whyfunction, 2019-01-11:
when I run "brew reinstall --build-from-source riscv-gcc", it says 'Error: No available formula with the name "riscv-gcc"'

sbeamer, 2019-01-13:
The riscv-gcc formula was renamed to riscv-gnu-toolchain.

Depending on the issues you were having, you may not need to rebuild from source since the bottles are relatively recent (rebuilt last month).