riscv-software-src/homebrew-riscv #16: --build-from-source does not work for riscv-gcc formula

dmzimmerman, 2018-03-01:
`brew reinstall --build-from-source riscv-gcc` does not complete; instead, it hangs at "Checking out branch master" (probably because the `riscv-gcc` repository has no `master` branch, and Homebrew defaults to checking out `master`).

dmzimmerman, 2018-03-01:
Actually, I believe I am mistaken here; it does complete, it just takes a very, very long time with no progress indication - the naming of the formula confused me into thinking that it was building from the `riscv-gcc` repository, when in fact it is building from the `riscv-gnu-toolchain` repository (which has many subrepositories, and takes a lot longer). It might be a reasonable idea to rename this formula to `riscv-gnu-toolchain` to avoid such confusion.

closed 2018-03-01 by dmzimmerman

sbeamer, 2018-03-02:
This is a reasonable suggestion, and there is no reason it has the name riscv-gcc other than it was the name of the repo before it was renamed to riscv-gnu-toolchain.

dmzimmerman, 2018-03-06:
Updating the reference in the documentation is good (thanks!), but probably renaming the actual formula to `riscv-gnu-toolchain` would be better as far as clarity going forward.