riscv-software-src/homebrew-riscv #17: Bottled riscv-gcc on High Sierra cannot create executables

dmzimmerman, 2018-03-01:
Attempting to compile a simple hello world with the High Sierra bottled riscv-gcc gives:

```
/usr/local/Cellar/riscv-gcc/gnu/bin/riscv64-unknown-elf-gcc -o hello hello.c
dyld: Library not loaded: /usr/local/opt/mpfr/lib/libmpfr.4.dylib
  Referenced from: /usr/local/Cellar/riscv-gcc/gnu/libexec/gcc/riscv64-unknown-elf/7.2.0/cc1
  Reason: image not found
riscv64-unknown-elf-gcc: internal compiler error: Abort trap: 6 (program cc1)
Please submit a full bug report,
with preprocessed source if appropriate.
See <https://gcc.gnu.org/bugs/> for instructions.
```

Looking in `/usr/local/opt/mpfr/lib`, it appears that there are a `libmpfr.dylib` and a `libmpfr.6.dylib` (as well as a `libmpfr.a`), but no `libmpfr.4.dylib`. Symlinking `libmpfr.6.dylib` to `libmpfr.4.dylib` allows gcc to work, but is clearly not a real solution to this problem.

dmzimmerman, 2018-03-01:
It appears that the Homebrew versions of libmpfr changed (from 3.x to 4.x) back in December, and that changed the library version from 4 to 6... since the bottle was linked against version 4, the bottle no longer works with up-to-date Homebrew packages. Rebuilding the formula from source fixed the problem (after about a half hour), so this is probably just a matter of replacing the bottled version with a current build.

sbeamer, 2018-03-02:
Yes, I believe this is the same issue as in #15. I will probably rebuild the bottles in time for the next RISC-V workshop, and that should fix the default behavior for most people.

sbeamer, 2018-03-03:
Ok, since Travis this issue breaks Travis, I just rebottled just riscv-gcc.

closed 2018-03-03 by sbeamer