riscv-software-src/homebrew-riscv #18: "brew test riscv-tools" fails

dmzimmerman, 2018-03-01:
The `brew test riscv-tools` test fails, because it tries to execute `riscv64-unknown-elf-gcc` from the shell's search path, when the `riscv-gcc` formula does not (on my system, at least) actually link it into `/usr/local` due to conflicts with my already-installed `binutils` package. There are, I think, two possible ways to fix this:

1) explicitly use the `Cellar` path to the `riscv-gcc` binary in the test, by replacing:

```
    system "riscv64-unknown-elf-gcc", (testpath/'hello.c')
```
with
```
    system "#{Formula["riscv-gcc"].opt_prefix}/bin/riscv64-unknown-elf-gcc", (testpath/'hello.c')
```
(I tried this, and it worked)

2) somehow resolve the conflicts that prevent Homebrew from linking the tools into `/usr/local` (I have not looked into this)

sbeamer, 2018-03-02:
I'm open to suggestion on what is the best homebrew practice for this. Looking at other brew formulas, they typically do something like `system "#{bin}/riscv64-unknown-elf-gcc"`, however, it doesn't  work in this case since riscv64-unknown-elf-gcc is made by another formula.

Your approach works on my system (which is still able to default to /usr/local). Could you open a PR with this fix?

sbeamer, 2018-03-06:
Thanks for submitting #20!

closed 2018-03-06 by sbeamer