riscv-software-src/homebrew-riscv #22: Include the 32 bit tools?

bradjc, 2018-07-26:
Is there any plan to add the 32 bit version of the tools (riscv32-unknown-elf) to the brew install?

sbeamer, 2018-10-10:
Please comment to re-open this if new info to contribute.

closed 2018-10-10 by sbeamer

bradjc, 2020-06-14:
Revisiting my old issue I totally forgot I made...

The [Tock](https://github.com/tock/tock) project is working on embedded OS support, and there is rapidly growing interest in RISC-V based platforms. The kernel uses LLVM, but we want to support GCC for userspace apps written in C. Long story short: it would be great for users to have easy access to a 32 bit RISC-V compiler on mac.

Right now, `brew install riscv-gnu-toolchain --with-multilib` works...but takes >30 min on my computer, which isn't very friendly for users. Ideally, there would be a pre-compiled option, since I don't want supporting RISC-V to feel like a burden.

I have three ideas, and I'm interested in what you think:
1. `--enable-multilib` could be set in the formula by default. I'm not actually sure why the toolchain isn't just riscv-unknown-elf-gcc that supports both 64 and 32 always...but I suppose that is a different topic.
2. Add a `riscv-gnu-toolchain-multilib.rb` formula.
3. Create `github.com/tock/homebrew-tock` with this hypothetical riscv-gnu-toolchain-multilib.rb formula. I don't think that forking this is good for the RISC-V ecosystem, but it is hypothetically possible.

Do you have a preference, or an idea I didn't think of?