riscv-software-src/homebrew-riscv #23: $RISCV is not set by this process. Can this step be automated or included in README?

BassMunkee, 2018-10-16:
The homebrew installer worked rather smoothly for me on Mac vs following the instructions on the risv-tools repo. For installing BOOM, the $RISCV environment variable needs to be set. I'm unsure what should that be after installing the tools using Homebrew. How do I find the path? Can this also be part of the README or automated?

BassMunkee, 2018-10-17:
I'm following the instructions for setting up riscv-boom by using the boom-template from here:

I installed the riscv-tools using homebrew and set $RISCV to /usr/local and added RISCV/bin to $PATH and was able to proceed forward. This allowed me to compile the verisim using make. 

But now if I try to run `make run-regression-tests` I get the error 
`make: *** No rule to make target `/usr/local/riscv64-unknown-elf/share/riscv-tests/isa/rv64ud-v-fcvt', needed by `output/rv64ud-v-fcvt'.  Stop.`

When I try to run `make run` I get the error
`make: *** No rule to make target `/usr/local/riscv64-unknown-elf/share/riscv-tests/isa/rv64um-v-mul', needed by `output/rv64um-v-mul'.  Stop.`

I know this is specific to the boom-template project but I believe this is happening because $RISCV environment variable is not set correctly as Homebrew downloads and compiles it in a different path from what is probably followed when manually compiling it following the instructions in those other projects. 

Any help/guidance is much appreciated.

BassMunkee, 2018-12-06:
> The homebrew installation does not set the $RISCV variable because it is not clear what to set it to. When manually installing riscv-tools, everything is installed into $RISCV. By contrast, with homebrew, each package is given its own separate directory. Thus setting $RISCV would be setting it for only one package.
>
> If you are in a usage scenario that expects $RISCV to be set, you might be best served by doing a manual installation. By doing so, you not only put all of the packages into the same directory, but you will also get the exact right versions of the tools (pointed to by the git submodules). This homebrew installation installs whatever version is in the most recent bottle, or head of master if you build from source.
>
> For working with boom-template (or rocket-chip), I would recommend performing a manual installation. To build the simulators, they will look at $RISCV to find riscv-fesvr, but to run tests it will be looking for riscv-tests inside of $RISCV.

Thanks for the clarification. Do you think something like this should go into the readme?  When I tried to set up Rocket I found the number of moving parts and the layers that need to be setup quite daunting. I also ran into some issues with the manual setup and then realised the option to use homebrew.

What I'm trying to say is that there are lot of folks who don't have much experience in open source / *nix tools who want to try using riscv and onboarding process can be made easier

sbeamer, 2018-12-06:
Admittedly, the README for rocket-chip is at times out of date or confusing, but I don't believe it recommends using brew for riscv-tools. Are you recommending revising the README in this repo or rocket-chip? I'm open to modifying the README in this repo, so please open a PR to show what you think would help reduce confusion.

BassMunkee, 2018-12-06:
Let me think this through and create a PR in the appropriate repo. Thanks again for your feedback

sbeamer, 2019-02-24:
Please re-open this issue to comment on it further.

closed 2019-02-24 by sbeamer