riscv-software-src/homebrew-riscv #29: riscv-gnu-toolchain make[12]: Nothing to be done for `install-data-am'.

SwordFaith, 2020-01-07:
run brew reinstall --build-from-source riscv-gnu-toolchain

==> ./configure --prefix=/usr/local/Cellar/riscv-gnu-toolchain/gnu
==> make
Last 15 lines from /Users/minds/Library/Logs/Homebrew/riscv-gnu-toolchain/02.make:
	  /bin/sh /private/tmp/riscv-gnu-toolchain-20200107-15367-2x9diz/riscv-gdb/gdb/gnulib/../../mkinstalldirs /usr/local/Cellar/riscv-gnu-toolchain/gnu/lib ; \
	fi ; \
	if test -f /usr/local/Cellar/riscv-gnu-toolchain/gnu/lib/charset.alias; then \
	  sed -f ref-add.sed /usr/local/Cellar/riscv-gnu-toolchain/gnu/lib/charset.alias > /usr/local/Cellar/riscv-gnu-toolchain/gnu/lib/charset.tmp ; \
	  /usr/bin/install -c -m 644 /usr/local/Cellar/riscv-gnu-toolchain/gnu/lib/charset.tmp /usr/local/Cellar/riscv-gnu-toolchain/gnu/lib/charset.alias ; \
	  rm -f /usr/local/Cellar/riscv-gnu-toolchain/gnu/lib/charset.tmp ; \
	else \
	  if $need_charset_alias; then \
	    sed -f ref-add.sed charset.alias > /usr/local/Cellar/riscv-gnu-toolchain/gnu/lib/charset.tmp ; \
	    /usr/bin/install -c -m 644 /usr/local/Cellar/riscv-gnu-toolchain/gnu/lib/charset.tmp /usr/local/Cellar/riscv-gnu-toolchain/gnu/lib/charset.alias ; \
	    rm -f /usr/local/Cellar/riscv-gnu-toolchain/gnu/lib/charset.tmp ; \
	  fi ; \
	fi
make[12]: Nothing to be done for `install-data-am'.
mkdir -p stamps/ && touch stamps/build-gdb-newlib

mac os x 10.15.2 
command line tools for xcode 11.3 Dec 3, 2019

run brew test riscv-tools report following errors:

dyld: Library not loaded: /usr/local/opt/isl/lib/libisl.21.dylib
  Referenced from: /usr/local/Cellar/riscv-gnu-toolchain/gnu/libexec/gcc/riscv64-unknown-elf/9.2.0/cc1
  Reason: image not found
riscv64-unknown-elf-gcc: internal compiler error: Abort trap: 6 signal terminated program cc1

sbeamer, 2020-01-14:
I haven't had a chance to test much on Catalina yet. Please let us know if you find a solution.

claford-v-lawrence, 2020-01-20:
> I haven't had a chance to test much on Catalina yet. Please let us know if you find a solution.

I have found a solution, current macOS version is 15.2

after installing, run the following command:

install_name_tool -change '/usr/local/opt/isl/lib/libisl.21.dylib' /usr/local/opt/isl/lib/libisl.dylib /usr/local/Cellar/riscv-gnu-toolchain/gnu/libexec/gcc/riscv64-unknown-elf/9.2.0/cc1

source:[stackoverflow](https://stackoverflow.com/questions/59538261/dyld-library-not-loaded-usr-local-opt-isl-lib-libisl-21-dylib)

after this, it runs successfully

hwj1995, 2020-02-14:
> I have found a solution, current macOS version is 15.2
>
> after installing, run the following command:
>
> install_name_tool -change '/usr/local/opt/isl/lib/libisl.21.dylib' [...]

Thank you very much! I don't solve this problem until finding your solution!

sbeamer, 2020-05-21:
Please reopen this if this is still an issue. Hopefully the bottles and fixes from 2 months ago fixed this.

closed 2020-05-21 by sbeamer