riscv-software-src/homebrew-riscv #30: Please rebuild the pre-compiled binary with isl@0.22

xuqingkuang, 2020-01-12:
The latest version of `isl` is [0.22](https://formulae.brew.sh/formula/isl#default), but the precompiled binary was linked to 0.21.

It breaks ```brew test riscv-tools```  process.

```shell
» brew test riscv-tools                                                      130 ↵
Testing riscv/riscv/riscv-tools
==> /usr/local/opt/riscv-gnu-toolchain/bin/riscv64-unknown-elf-gcc /private/tmp/riscv-tools-test-20200112-34522-1txdtl9/hello.c
Last 15 lines from /Users/xqkuang/Library/Logs/Homebrew/riscv-tools/test.01.riscv64-unknown-elf-gcc:
2020-01-12 12:56:38 +0800

/usr/local/opt/riscv-gnu-toolchain/bin/riscv64-unknown-elf-gcc
/private/tmp/riscv-tools-test-20200112-34522-1txdtl9/hello.c

dyld: Library not loaded: /usr/local/opt/isl/lib/libisl.21.dylib
  Referenced from: /usr/local/Cellar/riscv-gnu-toolchain/gnu/libexec/gcc/riscv64-unknown-elf/9.2.0/cc1
  Reason: image not found
riscv64-unknown-elf-gcc: internal compiler error: Abort trap: 6 signal terminated program cc1
Please submit a full bug report,
with preprocessed source if appropriate.
See <https://gcc.gnu.org/bugs/> for instructions.
Error: riscv/riscv/riscv-tools: failed
Failed executing:
/usr/local/Homebrew/Library/Homebrew/formula.rb:1932:in `block in system'
/usr/local/Homebrew/Library/Homebrew/formula.rb:1869:in `open'
/usr/local/Homebrew/Library/Homebrew/formula.rb:1869:in `system'
/usr/local/Homebrew/Library/Taps/riscv/homebrew-riscv/riscv-tools.rb:27:in `block in <class:RiscvTools>'
/usr/local/Homebrew/Library/Homebrew/formula.rb:1745:in `block (3 levels) in run_test'
/usr/local/Homebrew/Library/Homebrew/utils.rb:478:in `with_env'
/usr/local/Homebrew/Library/Homebrew/formula.rb:1744:in `block (2 levels) in run_test'
/usr/local/Homebrew/Library/Homebrew/formula.rb:865:in `with_logging'
/usr/local/Homebrew/Library/Homebrew/formula.rb:1743:in `block in run_test'
/usr/local/Homebrew/Library/Homebrew/formula.rb:1982:in `block in mktemp'
/usr/local/Homebrew/Library/Homebrew/mktemp.rb:57:in `block in run'
/usr/local/Homebrew/Library/Homebrew/mktemp.rb:57:in `chdir'
/usr/local/Homebrew/Library/Homebrew/mktemp.rb:57:in `run'
/usr/local/Homebrew/Library/Homebrew/formula.rb:1981:in `mktemp'
/usr/local/Homebrew/Library/Homebrew/formula.rb:1737:in `run_test'
/usr/local/Homebrew/Library/Homebrew/test.rb:40:in `block in <main>'
/usr/local/Homebrew/Library/Homebrew/vendor/portable-ruby/2.6.3/lib/ruby/2.6.0/timeout.rb:93:in `block in timeout'
/usr/local/Homebrew/Library/Homebrew/vendor/portable-ruby/2.6.3/lib/ruby/2.6.0/timeout.rb:33:in `block in catch'
/usr/local/Homebrew/Library/Homebrew/vendor/portable-ruby/2.6.3/lib/ruby/2.6.0/timeout.rb:33:in `catch'
/usr/local/Homebrew/Library/Homebrew/vendor/portable-ruby/2.6.3/lib/ruby/2.6.0/timeout.rb:33:in `catch'
/usr/local/Homebrew/Library/Homebrew/vendor/portable-ruby/2.6.3/lib/ruby/2.6.0/timeout.rb:108:in `timeout'
/usr/local/Homebrew/Library/Homebrew/test.rb:39:in `<main>'
```

Environment:
macOS: 10.14.5
Xcode: 11.3
 » gcc --version                                                                1 ↵
Configured with: --prefix=/Applications/Xcode.app/Contents/Developer/usr --with-gxx-include-dir=/Library/Developer/CommandLineTools/SDKs/MacOSX.sdk/usr/include/c++/4.2.1
Apple clang version 11.0.0 (clang-1100.0.33.16)
Target: x86_64-apple-darwin18.6.0
Thread model: posix
InstalledDir: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin

sbeamer, 2020-01-14:
I unfortunately upgraded my Mojave system to Catalina, so it will be difficult for me to build for Mojave.

In the meantime, you should try recompiling it yourself:
`brew reinstall --build-from-source riscv-gnu-toolchain`

xuqingkuang, 2020-01-15:
> `brew reinstall --build-from-source riscv-gnu-toolchain`

Sure thing.

closed 2020-01-15 by xuqingkuang