riscv-software-src/homebrew-riscv #31: formulae are missing version

josharian, 2020-03-04:
And as a result of recent homebrew changes, this causes brew operations on the risc-v tap to fail.

See https://github.com/Homebrew/homebrew-core/issues/51073 and https://github.com/Homebrew/brew/issues/7110 for details.

kevinfeng, 2020-03-09:
have the same issue here;  `brew tap riscv/riscv` will fail:
```
Updating Homebrew...
==> Tapping riscv/riscv
Cloning into '/usr/local/Homebrew/Library/Taps/riscv/homebrew-riscv'...
remote: Enumerating objects: 10, done.
remote: Counting objects: 100% (10/10), done.
remote: Compressing objects: 100% (6/6), done.
remote: Total 467 (delta 4), reused 10 (delta 4), pack-reused 457
Receiving objects: 100% (467/467), 65.97 KiB | 123.00 KiB/s, done.
Resolving deltas: 100% (289/289), done.
Error: Invalid formula: /usr/local/Homebrew/Library/Taps/riscv/homebrew-riscv/riscv-pk.rb
invalid attribute for formula 'riscv/riscv/riscv-pk': version (nil)
Error: Invalid formula: /usr/local/Homebrew/Library/Taps/riscv/homebrew-riscv/riscv-gnu-toolchain.rb
invalid attribute for formula 'riscv/riscv/riscv-gnu-toolchain': version (nil)
Error: Invalid formula: /usr/local/Homebrew/Library/Taps/riscv/homebrew-riscv/riscv-isa-sim.rb
invalid attribute for formula 'riscv/riscv/riscv-isa-sim': version (nil)
Error: Cannot tap riscv/riscv: invalid syntax in tap!
```

duwanqiebi, 2020-03-11:
have the same issue too

DagAgren, 2020-03-12:
Struggling with this here too.

duwanqiebi, 2020-03-13:
I have given up installing on mac and already installed on linux。

victoryang00, 2020-03-14:
just add `version "master"` to all of the `url` line solves the problem. the path is `/usr/local/Homebrew/Library/Taps/riscv/homebrew-riscv/riscv-pk.rb` you should git clone in this path

n-wen, 2020-03-14:
> just add `version "master"` to all of the `url` line solves the problem. the path is `/usr/local/Homebrew/Library/Taps/riscv/homebrew-riscv/riscv-pk.rb` you should git clone in this path

thanks a lot . I just folked this repo and added version to it, adding extra url parameter to <code> brew tap</code> just work.

> brew tap riscv/riscv https://github.com/n-wen/homebrew-riscv

PR #32 (2020-03-19): Add versions to formulas.
Homebrew now checks that version is set and will die if it is not.

Closes #31
closed 2020-03-19 by sbeamer