riscv-software-src/homebrew-riscv #34: Build faild in 10.13.4

zhayujie, 2020-05-16:
Errors occured when making riscv-isa-sim.

macOS: Hign Sierra 10.13.4
Xcode: 9.4.1

```
$ brew install riscv-tools

==> Installing riscv-tools from riscv/riscv
Warning: Your Xcode (9.4) is outdated.
Please update to Xcode 10.1 (or delete it).
Xcode can be updated from the App Store.

==> Installing dependencies for riscv/riscv/riscv-tools: riscv-isa-sim, readline, gawk, riscv-gnu-toolchain, riscv-gcc and riscv-pk
==> Installing riscv/riscv/riscv-tools dependency: riscv-isa-sim
Warning: Your Xcode (9.4) is outdated.
Please update to Xcode 10.1 (or delete it).
Xcode can be updated from the App Store.

==> Cloning https://github.com/riscv/riscv-isa-sim.git
Updating /Users/zyj/Library/Caches/Homebrew/riscv-isa-sim--git
==> Checking out branch master
Already on 'master'
Your branch is up to date with 'origin/master'.
HEAD is now at a210111 Merge pull request #461 from riscv/mstatus-spie-sie
==> ../configure --prefix=/usr/local/Cellar/riscv-isa-sim/master
==> make prefix=/usr/local
Last 15 lines from /Users/zyj/Library/Logs/Homebrew/riscv-isa-sim/02.make:
    fdt_parse_pmp_num((void *)dtb.c_str(), &pmp_num, "riscv");
    ^~~~~~~~~~~~~~~~~
../riscv/dts.h:18:5: note: candidate function not viable: no known conversion from 'reg_t *' (aka 'unsigned long long *') to 'unsigned long *' for 2nd argument
int fdt_parse_pmp_num(void *fdt, unsigned long *pmp_num,
    ^
../riscv/sim.cc:96:5: error: no matching function for call to 'fdt_parse_pmp_alignment'
    fdt_parse_pmp_alignment((void *)dtb.c_str(), &pmp_granularity, "riscv");
    ^~~~~~~~~~~~~~~~~~~~~~~
../riscv/dts.h:20:5: note: candidate function not viable: no known conversion from 'reg_t *' (aka 'unsigned long long *') to 'unsigned long *' for 2nd argument
int fdt_parse_pmp_alignment(void *fdt, unsigned long *pmp_align,
    ^
3 errors generated.
clang++ -MMD -MP  -DPREFIX=\"/usr/local\" -Wall -Wno-unused -g -O2 -std=c++11 -DTARGET_ARCH=""  -DTARGET_ARCH="" -I. -I.. -I../fesvr -I../riscv -I../dummy_rocc -I../fdt -I../softfloat -I../spike_main  -c ../riscv/mmu.cc
make: *** [sim.o] Error 1
make: *** Waiting for unfinished jobs....
```

sbeamer, 2020-05-18:
This is an issue with riscv-isa-sim and clang. I submitted a PR (https://github.com/riscv/riscv-isa-sim/pull/463), so hopefully this will be fixed soon.

aswaterman, 2020-05-18:
Sorry 'bout that.

zhayujie, 2020-05-19:
> This is an issue with riscv-isa-sim and clang. I submitted a PR ([riscv/riscv-isa-sim#463](https://github.com/riscv/riscv-isa-sim/pull/463)), so hopefully this will be fixed soon.
It works now, thanks!

zhayujie, 2020-05-19:
> Sorry 'bout that.

Thanks for your fix!

closed 2020-05-19 by zhayujie