riscv-software-src/homebrew-riscv #37: An strange error occurred when I run riscv64-unknown-elf-gdb

PKUFlyingPig, 2020-09-17:
I installed the riscv-toolchain on my Mac followed the instruction in the README, it works well with qemu but when I run the riscv64-unknown-elf-gdb, an error occurred as follows:
[image]
I cd into the /usr/local/Cellar/riscv-gnu-toolchain/master/share/ ,but don't find a file named gdb.
meantime, there is another error says
[image]
and each time I enter any gdb command e.g. step, stepi ...., this error repeats.
How to solve this problem ?

sbeamer, 2020-09-17:
We welcome contributions to add support for gdb. We currently do not support it.

Similar to https://github.com/riscv/homebrew-riscv/issues/26

closed 2020-09-17 by sbeamer

Kuluso97, 2021-03-04:
brew install gdb, and copy the /usr/local/Cellar/gdb/<version code>/share/gdb/python to your riscv destination directory shown in the warning. This works for me.