riscv-software-src/homebrew-riscv #39: gdb not work

isuchen-code, 2020-11-11:
Hi, 

i am currently work on xv6 riscv and it seems the gdb doesn't work. Could you please help me with it?

ERROR message:

➜  xv6-labs-2020 git:(pgtbl) ✗ riscv64-unknown-elf-gdb          
Python Exception <type 'exceptions.ImportError'> No module named gdb: 
riscv64-unknown-elf-gdb: warning: 
Could not load the Python gdb module from `/usr/local/Cellar/riscv-gnu-toolchain/master/share/gdb/python'.
Limited Python support is available from the _gdb module.
Suggest passing --data-directory=/path/to/gdb/data-directory.

VERSION: 
➜  xv6-labs-2020 git:(pgtbl) ✗ riscv64-unknown-elf-gcc --version
riscv64-unknown-elf-gcc (GCC) 10.1.0

➜  xv6-labs-2020 git:(pgtbl) ✗ riscv64-unknown-elf-gdb --version
Python Exception <type 'exceptions.ImportError'> No module named gdb: 
riscv64-unknown-elf-gdb: warning: 
Could not load the Python gdb module from `/usr/local/Cellar/riscv-gnu-toolchain/master/share/gdb/python'.
Limited Python support is available from the _gdb module.
Suggest passing --data-directory=/path/to/gdb/data-directory.
GNU gdb (GDB) 9.1

➜  xv6-labs-2020 git:(pgtbl) ✗ qemu-system-riscv64 --version
QEMU emulator version 5.1.0

PATH:
➜  xv6-labs-2020 git:(pgtbl) ✗ ls /usr/local/Cellar/riscv-gnu-toolchain/master/share 
gcc-10.1.0 info       man

no gdb/python in path

sbeamer, 2020-11-17:
Similar to #26 & #37. We have not implemented support for gdb, and we welcome PRs that add support for it.

closed 2020-11-17 by sbeamer

JinLingChristopher, 2020-12-03:
I just copy the `/usr/local/Cellar/gdb/10.1/share/gdb/python` to the folder, looks like the problem solved.