riscv-software-src/homebrew-riscv #42: Build failed in MacOS 10.14.5

TheColdVoid, 2020-12-31:
`/Users/thevoid/Library/Caches/Homebrew/riscv-gnu-toolchain--git/riscv-newlib
==> ./configure --prefix=/usr/local/Cellar/riscv-gnu-toolchain/master
==> make
Last 15 lines from /Users/thevoid/Library/Logs/Homebrew/riscv-gnu-toolchain/02.make:
if test "opcodes" = "gettext"; then \
	  if test -r /private/tmp/riscv-gnu-toolchain-20201231-21867-9b1gbx/riscv-gdb/opcodes/../mkinstalldirs; then \
	    /private/tmp/riscv-gnu-toolchain-20201231-21867-9b1gbx/riscv-gdb/opcodes/../mkinstalldirs /usr/local/Cellar/riscv-gnu-toolchain/master/share/gettext/po; \
	  else \
	    /private/tmp/riscv-gnu-toolchain-20201231-21867-9b1gbx/riscv-gdb/opcodes/mkinstalldirs /usr/local/Cellar/riscv-gnu-toolchain/master/share/gettext/po; \
	  fi; \
	  /usr/bin/install -c -m 644 /private/tmp/riscv-gnu-toolchain-20201231-21867-9b1gbx/riscv-gdb/opcodes/po/Makefile.in.in \
			  /usr/local/Cellar/riscv-gnu-toolchain/master/share/gettext/po/Makefile.in.in; \
	else \
	  : ; \
	fi
mkdir /usr/local/Cellar/riscv-gnu-toolchain/master/share/gdb/python/gdb/command
mkdir /usr/local/Cellar/riscv-gnu-toolchain/master/share/gdb/python/gdb/function
mkdir /usr/local/Cellar/riscv-gnu-toolchain/master/share/gdb/python/gdb/printer
mkdir -p stamps/ && touch stamps/build-gdb-newlib

If reporting this issue please do so at (not Homebrew/brew or Homebrew/core):
  https://github.com/riscv/homebrew-riscv/issues
`

TheColdVoid, 2021-01-15:
Is there anyone who can help with this problem?

TheColdVoid, 2021-01-15:
thevoid@TheVoidsMBP  ~  make --version
GNU Make 4.2.1
Built for x86_64-apple-darwin18.2.0
Copyright (C) 1988-2016 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.
 thevoid@TheVoidsMBP  ~  clang --version
Apple clang version 11.0.0 (clang-1100.0.33.12)
Target: x86_64-apple-darwin18.7.0
Thread model: posix
InstalledDir: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin
 thevoid@TheVoidsMBP  ~ 

sbeamer, 2021-02-07:
This seems to be a problem coming from riscv-gnu-toolchain which broke mac support (https://github.com/riscv/riscv-gnu-toolchain/pull/802). Hopefully a newer PR will restore it.

closed 2021-05-18 by sbeamer