riscv-software-src/homebrew-riscv #44: M1 Mac support

xhw1, 2021-01-18:
Hi, 

Is there a way to install `riscv-tools` on m1 Macs?
when I run `brew install riscv-tools` I got error:

```
==> Installing riscv/riscv/riscv-tools dependency: riscv-gnu-toolchain
		==> ./configure --prefix=/opt/homebrew/Cellar/riscv-gnu-toolchain/master
==> make
Last 15 lines from /Users/xiaohan/Library/Logs/Homebrew/riscv-gnu-toolchain/02.make:
      toplev::main(int, char**) in libbackend.a(toplev.o)
      gt_pch_save(__sFILE*) in libbackend.a(ggc-common.o)
      gt_pch_save(__sFILE*) in libbackend.a(ggc-common.o)
      gt_pch_restore(__sFILE*) in libbackend.a(ggc-common.o)
      gt_pch_restore(__sFILE*) in libbackend.a(ggc-common.o)
ld: symbol(s) not found for architecture arm64
ld: symbol(s) not found for architecture arm64
clang: error: linker command failed with exit code 1 (use -v to see invocation)
clang: error: linker command failed with exit code 1 (use -v to see invocation)
make[2]: *** [cc1] Error 1
make[2]: *** Waiting for unfinished jobs....
make[2]: *** [cc1plus] Error 1
rm gcov.pod fsf-funding.pod lto-dump.pod gpl.pod cpp.pod gfdl.pod gcc.pod gcov-dump.pod gcov-tool.pod
make[1]: *** [all-gcc] Error 2
make: *** [stamps/build-gcc-newlib-stage1] Error 2
```

GuangsZuo, 2021-01-26:
same issues, did you solve it ?

Grippy98, 2021-02-04:
Same error here... it's a linker problem but not there yet with a fix.

aswaterman, 2021-02-04:
See my pull request #48

Grippy98, 2021-02-04:
@aswaterman 
Nice, I got it to compile that way just fine. 

If you're feeling particularly bored/charitable, having the same issue over here: https://github.com/osx-cross/homebrew-avr/issues/220#issuecomment-773383065 

With AVR-GCC, I'm just not sure where the equivalent dir is to do sed in.

aswaterman, 2021-02-04:
Glad it worked. I will let someone else make the AVR change because I don’t have time to test it out today :(

sbeamer, 2021-02-07:
Will close this issue and suggest using #47 to discuss M1 support.

closed 2021-02-07 by sbeamer