riscv-software-src/homebrew-riscv #45: Suggest adding approximate size requirements in README

timmyjose, 2021-01-21:
I would recommend adding explicit size requirements in the README so that people can chooose upfront whether to install this formula or not. On my system (10.14.6), it started cloning the source code (as the README mentions correctly), but when it started eating GBs of hard disk, then I had to kill it. This information should be shared upfront so that people are aware of how much space (and time) they'll need, upfront.

sbeamer, 2021-02-07:
Warning added.

This is an unfortunate perfect storm. Homebrew aggressively recursively clones when building from git. riscv-gnu-toolchain includes quite a few submodules such as QEMU which become quite big when recursively cloned and it is not even used for this context. Will definitely appreciate if anyone has ideas on how to prevent Homebrew from doing recursive. Perhaps from zip and then manually init the needed submodules?

timmyjose, 2021-02-08:
> Warning added.
>
> This is an unfortunate perfect storm. Homebrew aggressively recursively clones when building from git. riscv-gnu-toolchain includes quite a few submodules such as QEMU which become [...]

That's sort of what I suspected as well. I don't know of a good all round solution for this either, but the adding the warning bit is still very much appreciated! :-)

closed 2021-02-08 by timmyjose