riscv-software-src/homebrew-riscv #51: Test failed on 11.2

zhongchengyong, 2021-02-27:
I have installed riscv-tools by brew install, then run
`brew test riscv-tools`
Error occurs as following:

`==> Testing riscv/riscv/riscv-tools
==> /usr/local/opt/riscv-gnu-toolchain/bin/riscv64-unknown-elf-gcc /private/tmp/riscv-tools-test-20210227-5616-1wuvb37/hello.c
Last 15 lines from /Users/zhongchengyong/Library/Logs/Homebrew/riscv-tools/test.01.riscv64-unknown-elf-gcc:
2021-02-27 09:51:14 +0800

/usr/local/opt/riscv-gnu-toolchain/bin/riscv64-unknown-elf-gcc
/private/tmp/riscv-tools-test-20210227-5616-1wuvb37/hello.c

dyld: Library not loaded: /usr/local/opt/isl/lib/libisl.22.dylib
  Referenced from: /usr/local/Cellar/riscv-gnu-toolchain/master/libexec/gcc/riscv64-unknown-elf/10.2.0/cc1
  Reason: image not found
riscv64-unknown-elf-gcc: internal compiler error: Abort trap: 6 signal terminated program cc1
Please submit a full bug report,
with preprocessed source if appropriate.
See <https://gcc.gnu.org/bugs/> for instructions.
Error: riscv/riscv/riscv-tools: failed
An exception occurred within a child process:
  BuildError: Failed executing: /usr/local/opt/riscv-gnu-toolchain/bin/riscv64-unknown-elf-gcc /private/tmp/riscv-tools-test-20210227-5616-1wuvb37/hello.c
/usr/local/Homebrew/Library/Homebrew/formula.rb:2096:in `block in system'
/usr/local/Homebrew/Library/Homebrew/formula.rb:2032:in `open'
/usr/local/Homebrew/Library/Homebrew/formula.rb:2032:in `system'
/usr/local/Homebrew/Library/Taps/riscv/homebrew-riscv/riscv-tools.rb:27:in `block in <class:RiscvTools>'
/usr/local/Homebrew/Library/Homebrew/formula.rb:1900:in `block (3 levels) in run_test'
/usr/local/Homebrew/Library/Homebrew/utils.rb:531:in `with_env'
/usr/local/Homebrew/Library/Homebrew/formula.rb:1899:in `block (2 levels) in run_test'
/usr/local/Homebrew/Library/Homebrew/formula.rb:904:in `with_logging'
/usr/local/Homebrew/Library/Homebrew/formula.rb:1898:in `block in run_test'
/usr/local/Homebrew/Library/Homebrew/mktemp.rb:63:in `block in run'
/usr/local/Homebrew/Library/Homebrew/mktemp.rb:63:in `chdir'
/usr/local/Homebrew/Library/Homebrew/mktemp.rb:63:in `run'
/usr/local/Homebrew/Library/Homebrew/formula.rb:2145:in `mktemp'
/usr/local/Homebrew/Library/Homebrew/formula.rb:1892:in `run_test'
/usr/local/Homebrew/Library/Homebrew/test.rb:43:in `block in <main>'
/usr/local/Homebrew/Library/Homebrew/vendor/portable-ruby/2.6.3_2/lib/ruby/2.6.0/timeout.rb:93:in `block in timeout'
/usr/local/Homebrew/Library/Homebrew/vendor/portable-ruby/2.6.3_2/lib/ruby/2.6.0/timeout.rb:33:in `block in catch'
/usr/local/Homebrew/Library/Homebrew/vendor/portable-ruby/2.6.3_2/lib/ruby/2.6.0/timeout.rb:33:in `catch'
/usr/local/Homebrew/Library/Homebrew/vendor/portable-ruby/2.6.3_2/lib/ruby/2.6.0/timeout.rb:33:in `catch'
/usr/local/Homebrew/Library/Homebrew/vendor/portable-ruby/2.6.3_2/lib/ruby/2.6.0/timeout.rb:108:in `timeout'
/usr/local/Homebrew/Library/Homebrew/test.rb:42:in `<main>'`

sbeamer, 2021-02-28:
I've pushed new bottles for Big Sur. Try reinstalling, and hopefully it will use the bottles and resolve your problem.

zhongchengyong, 2021-03-01:
> I've pushed new bottles for Big Sur. Try reinstalling, and hopefully it will use the bottles and resolve your problem.

It works, thank you very much.

closed 2021-03-01 by zhongchengyong