riscv-software-src/homebrew-riscv #53: Failed to install from homebrew

Turmiht, 2021-03-25:
After doing what readme asked me to do, I got error as follows when exec `brew install riscv-tools':
```
==> Installing riscv-tools from riscv/riscv
==> Downloading http://riscv.org.s3.amazonaws.com/bottles/riscv-gnu-toolchain-master.big_sur.bottle.
##O#- #
curl: (22) The requested URL returned error: 404 Not Found
Error: Failed to download resource "riscv-gnu-toolchain"
Download failed: http://riscv.org.s3.amazonaws.com/bottles/riscv-gnu-toolchain-master.big_sur.bottle.4.tar.gz
```
clicking the link returns raw XML data that means little to me.

sbeamer, 2021-03-25:
It appears the S3 storage from RISC-V we were using to host the bottles is unreachable. I'll follow up to find if something is down, path moved, or we will need to find a new place to host bottles.

VidocqH, 2021-03-26:
Has the same issue.👀

Turmiht, 2021-03-26:
Yep. I tried to build from source which proves out not that complex actually. Only caveat here is you need a case-sensitive partition on machine. Hope it helps someone before brew bottle works.

VidocqH, 2021-03-26:
Same issue

> Yep. I tried to build from source which proves out not that complex actually. Only caveat here is you need a case-sensitive partition on machine. Hope it helps someone before brew bottle works.

It helps, thanks!!!!!

cetola, 2021-03-26:
Should be resolved.

closed 2021-03-26 by cetola