riscv-software-src/homebrew-riscv #57: use cmodel=medany when multilib build is enabled

cfriedt, 2021-05-05:
Even when building the homebrew-riscv toolchain via the command below, there are a few different circumstances where certain relocations are unsupported.
```bash
brew install riscv-gnu-toolchain --enable-multilib
```

A number of tickets can be found here that highlight the various relocation errors:
https://github.com/riscv/riscv-gnu-toolchain/issues?q=is%3Aissue+relocation+%22--with-cmodel=medany%22

I propose using `--with-cmodel=medany` additionally when `--enable-multilib` is passed in to `brew`, as that solved the relocation issues for me.

There is a 1-liner fixup forthcoming.

cfriedt, 2021-05-05:
Examples of relocation errors of the sort described here can be found in riscv/riscv-gnu-toolchain#784 and riscv/riscv-gnu-toolchain#457

sbeamer, 2021-05-22:
Hopefully fixed by #59

closed 2021-05-22 by sbeamer