riscv-software-src/homebrew-riscv #60: sed: riscv-gcc/gcc/config.host: No such file or directory

GejinZ, 2021-06-05:
errors occur when `brew install riscv-tools`

[image]

I find this:
[image]

and my mac is not work on M1.

GejinZ, 2021-06-05:
And my macOS version is 10.15.7 (19H524)

ZonePG, 2021-06-18:
same error on my m1 macOS

ZonePG, 2021-06-18:
I solved it, just delete riscv-gcc and `brew install riscv-tools` again.

kalpesh2001, 2021-06-21:
Is there a plan to fix this issue? I tried to delete and reinstall with same error.

sbeamer, 2021-07-03:
This seems to be M1-specific, so I am unable to reproduce. Would be grateful for any community help.

closed 2022-07-17 by sbeamer