riscv-software-src/homebrew-riscv #61: riscv64-unknown-elf-gdb: Installation error: gdb._execute_unwinders function is missing

Happytreat, 2021-06-08:
Hi, I am currently work on xv6 riscv and it seems that riscv64-unknown-elf-gdb doesn't work. Could you please help me with it?

After running riscv64-unknown-elf-gdb, the ERROR message was: 

`Python Exception <type 'exceptions.NameError'> Installation error: gdb._execute_unwinders function is missing`

**.gdbinit**: (Expected to work) 
```
set confirm off
set architecture riscv:rv64
target remote 127.0.0.1:25501
symbol-file kernel/kernel
set disassemble-next-line auto
```

sbeamer, 2021-07-03:
How are you intending to use gdb in this context? In other words, what is emulating RISC-V on your mac?

jimizai, 2021-07-17:
I just copy the /usr/local/Cellar/gdb/10.1/share/gdb/python to the folder, looks like the problem solved.

closed 2021-08-01 by sbeamer

Smith-Cruise, 2021-08-16:
> I just copy the /usr/local/Cellar/gdb/10.1/share/gdb/python to the folder, looks like the problem solved.

Please tell me to copy to which folder, I'm facing the same problem you've met.

Virus-V, 2021-08-23:
```
cd /usr/local/Cellar/riscv-gnu-toolchain/master/share
mkdir gdb
cd gdb
cp -r /usr/local/Cellar/gdb/10.2/share/gdb/python .
```
@Smith-Cruise Executing these commands seems to solve my problem.