riscv-software-src/homebrew-riscv #64: M1 Mac Support for riscv64-unknown-linux-gnu- support

venkatakrishs, 2021-07-30:
I have currently installed the RISCV toolchain from homebrew-riscv repo and can find riscv64-unknown-elf-gdb riscv64-unknown-elf-gcc and all. But I cannot find riscv64-unknown-linux-gnu-gcc kind of elf which is used to compile linux. 

Any help on this is appreciated.

sbeamer, 2021-08-01:
This repo does not build the linux form of gcc (#50). We welcome PRs to add support for this. The main issue is a subtle issue with file naming when building glibc on a case-insensitive file system (default for mac).

venkatakrishs, 2021-08-03:
Then is it possible to install the GNU Linux toolchain in Rosetta 2 for M1 Mac.?

sbeamer, 2021-08-09:
The issue with the linux toolchain isn't M1/Rosetta, but linux's assumption of a case-sensitive file system, and by default, macOS uses a case-insensitive file system. The easiest way to work around this is to make a disk image for a case-sensitive file system and build the linux toolchain within it manually (without homebrew). Be sure to make the image big (>10 GB) so you do not run out of room. Once built, you can copy it out of the disk image and discard the image. These sorts of operations are difficult to put into a homebrew forumla.

closed 2021-08-11 by venkatakrishs