riscv-software-src/homebrew-riscv #67: brew test riscv-tools failed on macOS 11.3

ljerbi, 2021-08-30:
after installing risk-tools using brew, I get the following output when I run 
brew test riscv-tools

==> Testing riscv/riscv/riscv-tools
==> /opt/brew/opt/riscv-gnu-toolchain/bin/riscv64-unknown-elf-gcc /private/tmp/riscv-tools-test-20210830-24311-11gdeba/hello.c
Last 15 lines from /Users/lindajerbi/Library/Logs/Homebrew/riscv-tools/test.01.riscv64-unknown-elf-gcc:
2021-08-30 22:54:52 +0200

/opt/brew/opt/riscv-gnu-toolchain/bin/riscv64-unknown-elf-gcc
/private/tmp/riscv-tools-test-20210830-24311-11gdeba/hello.c

dyld: Library not loaded: /opt/brew/opt/gettext/lib/libintl.8.dylib
  Referenced from: /opt/brew/Cellar/riscv-gnu-toolchain/master/bin/../lib/gcc/riscv64-unknown-elf/11.1.0/../../../../riscv64-unknown-elf/bin/as
  Reason: image not found
riscv64-unknown-elf-gcc: internal compiler error: Abort trap: 6 signal terminated program as
Please submit a full bug report,
with preprocessed source if appropriate.
See <https://gcc.gnu.org/bugs/> for instructions.
Error: riscv/riscv/riscv-tools: failed
An exception occurred within a child process:
  BuildError: Failed executing: /opt/brew/opt/riscv-gnu-toolchain/bin/riscv64-unknown-elf-gcc /private/tmp/riscv-tools-test-20210830-24311-11gdeba/hello.c
/opt/brew/Library/Homebrew/formula.rb:2137:in `block in system'
/opt/brew/Library/Homebrew/formula.rb:2073:in `open'
/opt/brew/Library/Homebrew/formula.rb:2073:in `system'
/opt/brew/Library/Taps/riscv/homebrew-riscv/riscv-tools.rb:26:in `block in <class:RiscvTools>'
/opt/brew/Library/Homebrew/formula.rb:1937:in `block (3 levels) in run_test'
/opt/brew/Library/Homebrew/utils.rb:550:in `with_env'
/opt/brew/Library/Homebrew/formula.rb:1936:in `block (2 levels) in run_test'
/opt/brew/Library/Homebrew/formula.rb:918:in `with_logging'
/opt/brew/Library/Homebrew/formula.rb:1935:in `block in run_test'
/opt/brew/Library/Homebrew/mktemp.rb:63:in `block in run'
/opt/brew/Library/Homebrew/mktemp.rb:63:in `chdir'
/opt/brew/Library/Homebrew/mktemp.rb:63:in `run'
/opt/brew/Library/Homebrew/formula.rb:2186:in `mktemp'
/opt/brew/Library/Homebrew/formula.rb:1929:in `run_test'
/opt/brew/Library/Homebrew/test.rb:43:in `block in <main>'
/System/Library/Frameworks/Ruby.framework/Versions/2.6/usr/lib/ruby/2.6.0/timeout.rb:93:in `block in timeout'
/System/Library/Frameworks/Ruby.framework/Versions/2.6/usr/lib/ruby/2.6.0/timeout.rb:33:in `block in catch'
/System/Library/Frameworks/Ruby.framework/Versions/2.6/usr/lib/ruby/2.6.0/timeout.rb:33:in `catch'
/System/Library/Frameworks/Ruby.framework/Versions/2.6/usr/lib/ruby/2.6.0/timeout.rb:33:in `catch'
/System/Library/Frameworks/Ruby.framework/Versions/2.6/usr/lib/ruby/2.6.0/timeout.rb:108:in `timeout'
/opt/brew/Library/Homebrew/test.rb:48:in `<main>'

sbeamer, 2021-08-31:
I might try rebuilding the tools from source (with the `--build-from-source`) flag. It appears you moved your brew install location, which is normally fine, but sometimes it can break bottles (pre-compiled installs).

zqy1018, 2021-09-04:
I also ran into this problem on my Apple Silicon Mac (with macOS 11.4). I guess that there is something incompatible with the ARM architecture, and thus I re-installed the toolchain with another `brew`(previously installed with `arch -x86_64` in Rosseta 2 environment). This seems to work.

closed 2021-12-17 by sbeamer