riscv-software-src/homebrew-riscv #69: Installation error with MacOS Catalina (Xcode is not present)

gtangg12, 2021-09-09:
==> Installing riscv-tools from riscv/riscv
==> Installing dependencies for riscv/riscv/riscv-tools: riscv-pk
==> Installing riscv/riscv/riscv-tools dependency: riscv-pk
==> ../configure --prefix=/usr/local/Cellar/riscv-pk/master --host=riscv64-unknown-elf
Last 15 lines from /Users/gtangg12/Library/Logs/Homebrew/riscv-pk/01.configure:
2021-09-09 17:28:31 -0400

../configure
--prefix=/usr/local/Cellar/riscv-pk/master
--host=riscv64-unknown-elf

checking build system type... x86_64-apple-darwin19.6.0
checking host system type... riscv64-unknown-elf
checking for riscv64-unknown-elf-gcc... riscv64-unknown-elf-gcc
checking whether the C compiler works... no
configure: error: in `/private/tmp/riscv-pk-20210909-67644-b7axet/build':
configure: error: C compiler cannot create executables
See `config.log' for more details

If reporting this issue please do so at (not Homebrew/brew or Homebrew/core):
  https://github.com/riscv/homebrew-riscv/issues

Traceback (most recent call last):
	28: from /usr/local/Homebrew/Library/Homebrew/build.rb:229:in `<main>'
	27: from /usr/local/Homebrew/Library/Homebrew/build.rb:133:in `install'
	26: from /usr/local/Homebrew/Library/Homebrew/utils.rb:550:in `with_env'
	25: from /usr/local/Homebrew/Library/Homebrew/build.rb:138:in `block in install'
	24: from /usr/local/Homebrew/Library/Homebrew/formula.rb:1275:in `brew'
	23: from /usr/local/Homebrew/Library/Homebrew/formula.rb:2434:in `stage'
	22: from /usr/local/Homebrew/Library/Homebrew/vendor/portable-ruby/2.6.3_2/lib/ruby/2.6.0/forwardable.rb:230:in `stage'
	21: from /usr/local/Homebrew/Library/Homebrew/resource.rb:91:in `stage'
	20: from /usr/local/Homebrew/Library/Homebrew/resource.rb:116:in `unpack'
	19: from /usr/local/Homebrew/Library/Homebrew/resource.rb:199:in `mktemp'
	18: from /usr/local/Homebrew/Library/Homebrew/mktemp.rb:63:in `run'
	17: from /usr/local/Homebrew/Library/Homebrew/mktemp.rb:63:in `chdir'
	16: from /usr/local/Homebrew/Library/Homebrew/mktemp.rb:63:in `block in run'
	15: from /usr/local/Homebrew/Library/Homebrew/resource.rb:117:in `block in unpack'
	14: from /usr/local/Homebrew/Library/Homebrew/download_strategy.rb:102:in `stage'
	13: from /usr/local/Homebrew/Library/Homebrew/download_strategy.rb:110:in `chdir'
	12: from /usr/local/Homebrew/Library/Homebrew/resource.rb:121:in `block (2 levels) in unpack'
	11: from /usr/local/Homebrew/Library/Homebrew/formula.rb:2454:in `block in stage'
	10: from /usr/local/Homebrew/Library/Homebrew/utils.rb:550:in `with_env'
	 9: from /usr/local/Homebrew/Library/Homebrew/formula.rb:2455:in `block (2 levels) in stage'
	 8: from /usr/local/Homebrew/Library/Homebrew/formula.rb:1282:in `block in brew'
	 7: from /usr/local/Homebrew/Library/Homebrew/build.rb:178:in `block (2 levels) in install'
	 6: from /usr/local/Homebrew/Library/Taps/riscv/homebrew-riscv/riscv-pk.rb:22:in `install'
	 5: from /usr/local/Homebrew/Library/Homebrew/vendor/portable-ruby/2.6.3_2/lib/ruby/2.6.0/fileutils.rb:128:in `cd'
	 4: from /usr/local/Homebrew/Library/Homebrew/vendor/portable-ruby/2.6.3_2/lib/ruby/2.6.0/fileutils.rb:128:in `chdir'
	 3: from /usr/local/Homebrew/Library/Taps/riscv/homebrew-riscv/riscv-pk.rb:23:in `block in install'
	 2: from /usr/local/Homebrew/Library/Homebrew/formula.rb:2225:in `system'
	 1: from /usr/local/Homebrew/Library/Homebrew/formula.rb:2225:in `open'
/usr/local/Homebrew/Library/Homebrew/formula.rb:2289:in `block in system': Failed executing: ../configure --prefix=/usr/local/Cellar/riscv-pk/master --host=riscv64-unknown-elf (BuildError)
	9: from /usr/local/Homebrew/Library/Homebrew/brew.rb:162:in `<main>'
	8: from /usr/local/Homebrew/Library/Homebrew/brew.rb:174:in `rescue in <main>'
	7: from /usr/local/Homebrew/Library/Homebrew/exceptions.rb:509:in `dump'
	6: from /usr/local/Homebrew/Library/Homebrew/exceptions.rb:455:in `issues'
	5: from /usr/local/Homebrew/Library/Homebrew/exceptions.rb:459:in `fetch_issues'
	4: from /usr/local/Homebrew/Library/Homebrew/utils/github.rb:61:in `issues_for_formula'
	3: from /usr/local/Homebrew/Library/Homebrew/utils/github.rb:30:in `search_issues'
	2: from /usr/local/Homebrew/Library/Homebrew/utils/github.rb:160:in `search'
	1: from /usr/local/Homebrew/Library/Homebrew/utils/github/api.rb:234:in `open_rest'
/usr/local/Homebrew/Library/Homebrew/utils/github/api.rb:304:in `raise_error': Validation Failed: [{"message"=>"The listed users and repositories cannot be searched either because the resources do not exist or you do not have permission to view them.", "resource"=>"Search", "field"=>"q", "code"=>"invalid"}] (GitHub::API::ValidationFailedError)

gtangg12, 2021-09-09:
x86 Intel CPU

tuupola, 2021-09-12:
Do you have XCode Command Line tools installed?

```shell
$ xcode-select --print-path 
/Library/Developer/CommandLineTools
```

sbeamer, 2021-09-15:
I'm a little stumped. Did the compiler (riscv-gnu-toolchain) install smoothly already?

The compiler should be installed before attempting to build riscv-pk. Based on the excerpt, it appears configure while preparing to build riscv-pk may have gotten confused about which compiler to use.

gtangg12, 2021-09-15:
Yea I have XCode Command Line tools. Not sure what exactly is going on; I just got it to work with docker (Ubuntu).

closed 2021-09-15 by gtangg12