riscv-software-src/homebrew-riscv #72: Build failed on macOS Catalina 10.15.5

Hope-of-countryside, 2021-10-06:
I build riscv through homebrew by default gcc(in clang) and gcc11.
``` 
brew install riscv-tools
brew install --cc=gcc-11 riscv-tools
```
But I failed, and infos show like this. Can anyone tells me what can I do?

```
==> ../configure --prefix=/usr/local/Cellar/riscv-isa-sim/master --with-target="
==> make prefix=/usr/local
Last 15 lines from /Users/zhangyixiao/Library/Logs/Homebrew/riscv-isa-sim/02.make:
  "state_t::num_triggers", referenced from:
      state_t::reset(processor_t*, unsigned long long) in libriscv.a(processor.o)
Undefined symbols for architecture x86_64:
  "state_t::num_triggers", referenced from:
      state_t::reset(processor_t*, unsigned long long) in libriscv.a(processor.o)
ld: symbol(s) not found for architecture x86_64
ld: symbol(s) not found for architecture x86_64
ld: symbol(s) not found for architecture x86_64
clang: error: linker command failed with exit code 1 (use -v to see invocation)
make: *** [spike-log-parser] Error 1
make: *** Waiting for unfinished jobs....
clang: error: linker command failed with exit code 1 (use -v to see invocation)
make: *** [spike] Error 1
clang: error: linker command failed with exit code 1 (use -v to see invocation)
make: *** [libcustomext.so] Error 1

If reporting this issue please do so at (not Homebrew/brew or Homebrew/core):
  https://github.com/riscv/homebrew-riscv/issues

Traceback (most recent call last):
	28: from /usr/local/Homebrew/Library/Homebrew/build.rb:220:in `<main>'
	27: from /usr/local/Homebrew/Library/Homebrew/build.rb:124:in `install'
	26: from /usr/local/Homebrew/Library/Homebrew/utils.rb:550:in `with_env'
	25: from /usr/local/Homebrew/Library/Homebrew/build.rb:129:in `block in install'
	24: from /usr/local/Homebrew/Library/Homebrew/formula.rb:1284:in `brew'
	23: from /usr/local/Homebrew/Library/Homebrew/formula.rb:2442:in `stage'
	22: from /usr/local/Homebrew/Library/Homebrew/vendor/portable-ruby/2.6.3_2/lib/ruby/2.6.0/forwardable.rb:230:in `stage'
	21: from /usr/local/Homebrew/Library/Homebrew/resource.rb:96:in `stage'
	20: from /usr/local/Homebrew/Library/Homebrew/resource.rb:121:in `unpack'
	19: from /usr/local/Homebrew/Library/Homebrew/resource.rb:208:in `mktemp'
	18: from /usr/local/Homebrew/Library/Homebrew/mktemp.rb:63:in `run'
	17: from /usr/local/Homebrew/Library/Homebrew/mktemp.rb:63:in `chdir'
	16: from /usr/local/Homebrew/Library/Homebrew/mktemp.rb:63:in `block in run'
	15: from /usr/local/Homebrew/Library/Homebrew/resource.rb:122:in `block in unpack'
	14: from /usr/local/Homebrew/Library/Homebrew/download_strategy.rb:102:in `stage'
	13: from /usr/local/Homebrew/Library/Homebrew/download_strategy.rb:110:in `chdir'
	12: from /usr/local/Homebrew/Library/Homebrew/resource.rb:126:in `block (2 levels) in unpack'
	11: from /usr/local/Homebrew/Library/Homebrew/formula.rb:2462:in `block in stage'
	10: from /usr/local/Homebrew/Library/Homebrew/utils.rb:550:in `with_env'
	 9: from /usr/local/Homebrew/Library/Homebrew/formula.rb:2463:in `block (2 levels) in stage'
	 8: from /usr/local/Homebrew/Library/Homebrew/formula.rb:1291:in `block in brew'
	 7: from /usr/local/Homebrew/Library/Homebrew/build.rb:169:in `block (2 levels) in install'
	 6: from /usr/local/Homebrew/Library/Taps/riscv/homebrew-riscv/riscv-isa-sim.rb:17:in `install'
	 5: from /usr/local/Homebrew/Library/Homebrew/vendor/portable-ruby/2.6.3_2/lib/ruby/2.6.0/fileutils.rb:128:in `cd'
	 4: from /usr/local/Homebrew/Library/Homebrew/vendor/portable-ruby/2.6.3_2/lib/ruby/2.6.0/fileutils.rb:128:in `chdir'
	 3: from /usr/local/Homebrew/Library/Taps/riscv/homebrew-riscv/riscv-isa-sim.rb:20:in `block in install'
	 2: from /usr/local/Homebrew/Library/Homebrew/formula.rb:2233:in `system'
	 1: from /usr/local/Homebrew/Library/Homebrew/formula.rb:2233:in `open'
/usr/local/Homebrew/Library/Homebrew/formula.rb:2297:in `block in system': Failed executing: make prefix=/usr/local (BuildError)
	9: from /usr/local/Homebrew/Library/Homebrew/brew.rb:162:in `<main>'
	8: from /usr/local/Homebrew/Library/Homebrew/brew.rb:174:in `rescue in <main>'
	7: from /usr/local/Homebrew/Library/Homebrew/exceptions.rb:511:in `dump'
	6: from /usr/local/Homebrew/Library/Homebrew/exceptions.rb:457:in `issues'
	5: from /usr/local/Homebrew/Library/Homebrew/exceptions.rb:461:in `fetch_issues'
	4: from /usr/local/Homebrew/Library/Homebrew/utils/github.rb:61:in `issues_for_formula'
	3: from /usr/local/Homebrew/Library/Homebrew/utils/github.rb:30:in `search_issues'
	2: from /usr/local/Homebrew/Library/Homebrew/utils/github.rb:160:in `search'
	1: from /usr/local/Homebrew/Library/Homebrew/utils/github/api.rb:234:in `open_rest'
/usr/local/Homebrew/Library/Homebrew/utils/github/api.rb:304:in `raise_error': Validation Failed: [{"message"=>"The listed users and repositories cannot be searched either because the resources do not exist or you do not have permission to view them.", "resource"=>"Search", "field"=>"q", "code"=>"invalid"}] (GitHub::API::ValidationFailedError)
```

sbeamer, 2021-12-27:
We aren't able to reproduce this. Does the issue still persist?

Hope-of-countryside, 2021-12-28:
> 

> We aren't able to reproduce this. Does the issue still persist?

Hope-of-countryside, 2021-12-28:
> We aren't able to reproduce this. Does the issue still persist?

I fix it by update my macOS to 12.0.1. thx~

closed 2021-12-28 by Hope-of-countryside