riscv-software-src/homebrew-riscv #75: V extension

liangkaiwang, 2021-10-31:
Does the current version support the vector extension? 

I try running this:

 riscv64-unknown-elf-gcc main.c asm_func.s -o ./a.out -march=rv64gcv
and getting this:

Assembler messages:
Error: cannot find default versions of the ISA extension `v'
/var/folders/qk/896_3mtn7ls3t09tdm0lr2jr0000gn/T//ccIFKgs2.s:3: Error: cannot find default versions of the ISA extension `v'
Assembler messages:
Error: cannot find default versions of the ISA extension `v'

GPSBabelDeveloper, 2021-11-22:
GCC trunk doesn't support draft extensions. V is in the very final ("weeks"? There's a big RISC-V conference in early December that may help push things along) stages of support, but it's not final so it can't go in GCC trunk.

There are various forks in vendor chains (sifive, allwinner/alibaba) that support some given draft of V, but it's kind of a messy since there is no silicon at this time that supports V because a ratified version of V doesn't exist.

SiFive produces Mac binaries for their cores, but the V support is mostly for their own customers doing silicon work with their cores or on a simulator. See https://lists.riscv.org/g/tech-vector-ext/message/364 

Since the various drafts of V are incompatible with each other, prepare to be meticulous about matching whatever V support is produced by whatever branch you find with whatever V support is supported by your emulator or whatever pre-release version is supported by your use case.
Nezha/C906 is about the only volume device with V at all and it's draft version 0.7.1. Andes and CSky also have their own branches for whatever vendor extensions (it's not official RISC-V V since that doesn't exist yet as of this writing)  that you may need to chase down. They are, of course, somewhat incompatible with each other.

It's probably still too much of a wild west for Homebrew. Someday, 1.0 will be passed and it'll get integrated into GCC upstream. See Jim's comments in https://lists.riscv.org/g/tech-vector-ext/topic/74053961?p=Created,,,20,2,0,0 for one of the many discussions on how messy this situation is. Until then, integration really is left to those working with prerelease drafts.

sbeamer, 2021-12-17:
Thank you @GPSBabelDeveloper for your thorough and very correct comment!

Yes, the homebrew only pulls from the riscv-gnu-toolchain repo, and even that sometimes doesn't quite keep up anymore. Perhaps it is time to build from gcc proper such as #36.

GPSBabelDeveloper, 2021-12-17:
You're welcome. Keeping all the parts of the chain in sync has been a problem since I first started messing with GNU chain builds in the late 80's.  On some obscure target, GCC does something AS doesn't like. On another, AS does somethng LD doesn't like. On a third, LD does something GDB doesn't like, etc... (We had this close to home about this time last year when I was the first one that tried this tap on M1 and we got into trouble with host OS issues, while a relevant patch had been upstreamed for a while, but we were too old for that change to work so we just dropped a big hammer on it.)
aaiauwwwwwwwu                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                         [... cut ...] .................................................................N/.M
My *guess* is that people doing RISC-V work on a Mac are probably capable of building their own packages and doing the integration themselves. Slotting in development builds of any of them individually would be dangerous, but building the four I mentioned earlier (I don't use their libc, so I can't speak there) from release trees independently rom ddddjkjjjjk and then overlaying the install  either via a different --exec-prefi

victoryang00, 2021-12-23:
See my [homebrew-riscv](https://github.com/victoryang00/homebrew-riscv).

GPSBabelDeveloper, 2021-12-23:
Suggest adding WHICH version of Vector is supported since the only chip you
can buy (I think still true) is the pre-ratification D1 family with 0.7.1
but plenty of people are working with simulators or FPGA and such with 1.0.

hossein1387, 2022-02-08:
> 

I installed `riscv-gnu-toolchain` using ` brew reinstall --build-from-source riscv-gnu-toolchain` and I was hoping to be able to compile for a vector machine. However, It seems the command above does not enable the vector extension and I am keep getting errors like this:

`Error: cannot find default versions of the ISA extension v`

Here is my build command:

`
/opt/homebrew/bin/riscv64-unknown-elf-gcc -mcmodel=medany -march=rv64gcv -mabi=lp64d -static -std=gnu99 -O3 -ffast-math -fdiagnostics-color=always -g simple_vector_add.c -o simple_vector_add
`

Any suggestions?

GPSBabelDeveloper, 2022-02-08:
Gcc vector support has been depriorutized. Use LLVM if you need V,
especially auto vectorization.

hossein1387, 2022-02-08:
Thanks for your answer, Is there any homebrew tap that I can use? I tried to compile LLVM with vector extension using instructions provided in Ara: https://github.com/pulp-platform/ara
it worked perfectly on a linux machine but on my Mac, it fails when it tries to compile `toolchain-llvm-rt`: https://github.com/pulp-platform/ara/blob/ede4c58bf6a4153b0adba7f3561800383e9b4ca8/Makefile#L95

GPSBabelDeveloper, 2022-02-08:
I'm not familiar and you didn't include errors or context.

I'd never heard of Ara, but V. 10 is positively ancient and it's likely
that nothing contemporary supports that. There's already a big rift between
0.7.1 (shipping in one chip) and the ratified (in December) 1.0 which isn't
on any chip you can buy today.

hossein1387, 2022-02-08:
Thanks again for your reply. Here is the error log when I try to compile `llvm-rt`:
[error_llvm_rt.log](https://github.com/riscv-software-src/homebrew-riscv/files/8020378/error_llvm_rt.log)

Keep in mind that r Ara is a 64-bit Vector Unit, compatible with the RISC-V Vector Extension Version 0.10. And again, I am able to compile `llvm` and `llvm-rt` on a linux machine but I get the above error when I try to compile it on my Mac (I have a Mac with M1 chip).

GPSBabelDeveloper, 2022-02-08:
It looks like someone/something isn't picking up int_least8_t, which should
be defined in <stdint.h> Try including that or find out why it's not being
included. It looks like a common problem
in compiler-rt/lib/builtins/atomic*.  so it's probably inside some common
header that's included by the atomics code in that directory. If you can't
work it out, head to https://llvm.org/docs/HowToSubmitABug.html; I don't
know if IRC, Mailing Lists, or heading straight to the bug tracker works
best for this.

But let me reiterate: even if you get it building, the development trunk
is probably not going to actually generate code for V 0.10. I think they
removed code for 0.7.1, which is actually shipping on a somewhat popular
chip in favor of supporting only the ratified V 1.0. It seems extremely
unlikely there's support for 0.10 draft. You may have to look for an old
version of LLVM to support your old version of RVV, if it was ever
supported at all. Most of the development teams have taken the stance of
not supporting any of the draft versions of anything in the long term as
it's a support nightmare. This leads to the super weird situationi right
now that things in the real world (VLC did it this week) are supporting
RISC-V extensions that exist only virtually or in ASIC/emulation instead of
in "real" chips on developer boards.

Successfully building a compiler and having a compiler that emits opcodes
and patterns appropriate for your target are two somewhat related things.
The second depends on the first. With minor changes, you should be able to
get it working on MacOS as well as you do on Linux. What you're showing
probably comes down to a missing #define or an #include or something
finding a native header instead of a cross header or something. It's likely
to be minor when you find it.

Good luck. Your best help is likely to come from within the LLVM community.

victoryang00, 2022-02-08:
For D1 board, you could just download [T-head toolchains](https://occ.t-head.cn/community/download?id=3913221581316624384) with options `-march=ck806`, they have other modifications to the spec other than VExt.

GPSBabelDeveloper, 2022-02-08:
D1 (and the rather more rare C910 chip that's also based on the T-head
cores) implements 0.7.1.

Which of those builds is appropriate for MacOS? Because the site seems to
require registration or some kind of password. Even with Google Translate,
I can't coorce enough English out of it to penetrate.

If the Homebrew team is willing to take on the support load of what's
essentially a dead vendor-specific implementation of the chain, it would be
good for someone with need/fluency in such things to help spin up a build
for a tap. Someone probably needs to answer how to get support for a
vendor-proprietary toolchain...in a language understood by the maintainer.
Maybe that volunteer is you.

You don't have to read too much between my words to see that I'm not
volunteering to do so, but I know that D1 is only going to continue to sell
no matter how distasteful I find Allwinner's handling of this part.
T-head/Allwinner should upstream these changes to improve their
visibility/supportability, IMO, but I respect that wasn't the question
asked. :-)

GPSBabelDeveloper, 2022-02-10:
I'm reluctant to come back to this after I've already said this is the
wrong place to ask about something that probably won't meet your
expectations anyway, but since just had to spin through an LLVM build for
unrelated reasons, I just wanted to report back that the LLVM trunk
successfully builds on MacOS (uname -r: 21.3.0) and emits perfectly lovely
RVV 1.0 code.

Using the config recipe from https://github.com/sifive/riscv-llvm, my M1
chomped through all 3300-ish compiles with merely one "variable not used"
warning and crafted an LLVM. When I then use the source from
https://godbolt.org/z/xad7fn8Gz, I get the same vector code that Godbolt
generates from clang13 with the pertinent part being:
my_memcpy:
        beqz    a2, .LBB1_2
.LBB1_1:
        vsetvli a3, a2, e8, m4, ta, mu
        vle8.v  v8, (a1)
        vse8.v  v8, (a0)
        sub     a2, a2, a3
        add     a1, a1, a3
        add     a0, a0, a3
        bnez    a2, .LBB1_1
.LBB1_2:
        ret
.Lfunc_end1:
        .size   my_memcpy, .Lfunc_end1-my_memcpy

        .ident  "clang version 15.0.0 (
https://github.com/llvm/llvm-project.git
7d7ebf3cf06e6845dc5b83912f0adecbf69e1123)"

So we're way off topic for this report, but Clang trunk DOES successfully
build and generate RVV  code for me.

Good luck in finding the difference for your case.

rkabrick, 2022-05-24:
Hello! Sorry to resurrect this... Just wanted to clarify:

Is there a homebrew command that will use this tap in conjunction with the LLVM trunk for V-extension support? I apologize if this is a dumb question, I am not at all familiar with homebrew beyond `brew install`

victoryang00, 2022-05-26:
> Hello! Sorry to resurrect this... Just wanted to clarify:
> 
> Is there a homebrew command that will use this tap in conjunction with the LLVM trunk for V-extension support? I apologize if this is a dumb question, I am not at all familiar with homebrew beyond `brew install`

No, but I could make a homebrew formula using https://chocopy-llvm.github.io/chocopy-llvm/#/common/build.

rkabrick, 2022-05-26:
If it’s not too much trouble that would be fantastic! :)

victoryang00, 2022-05-27:
> If it’s not too much trouble that would be fantastic! :)
> […](#)

checkout https://github.com/victoryang00/homebrew-riscv