riscv-software-src/homebrew-riscv #76: M1 pro brew install

dingzhaohan, 2021-11-01:


dingzhaohan, 2021-11-01:
==> Installing riscv-tools from riscv/riscv
==> Installing dependencies for riscv/riscv/riscv-tools: riscv-gnu-toolchain, dtc, riscv-isa-sim and riscv-pk
==> Installing riscv/riscv/riscv-tools dependency: riscv-gnu-toolchain
==> sed -i .bak s/.*=host-darwin.o$// riscv-gcc/gcc/config.host
==> sed -i .bak s/.* x-darwin.$// riscv-gcc/gcc/config.host
==> ./configure --prefix=/opt/homebrew/Cellar/riscv-gnu-toolchain/master --with-cmodel=medany --enable-multilib
==> make
Last 15 lines from /Users/dingzhaohan/Library/Logs/Homebrew/riscv-gnu-toolchain/04.make:
installing /private/tmp/riscv-gnu-toolchain-20211101-36259-f4b17p/riscv-gdb/opcodes/po/vi.gmo as /opt/homebrew/Cellar/riscv-gnu-toolchain/master/share/locale/vi/LC_MESSAGES/opcodes.mo
install: /private/tmp/riscv-gnu-toolchain-20211101-36259-f4b17p/riscv-gdb/opcodes/po/zh_CN.gmo: No such file or directory
installing /private/tmp/riscv-gnu-toolchain-20211101-36259-f4b17p/riscv-gdb/opcodes/po/zh_CN.gmo as /opt/homebrew/Cellar/riscv-gnu-toolchain/master/share/locale/zh_CN/LC_MESSAGES/opcodes.mo
if test "opcodes" = "gettext"; then \
	  if test -r /private/tmp/riscv-gnu-toolchain-20211101-36259-f4b17p/riscv-gdb/opcodes/../mkinstalldirs; then \
	    /private/tmp/riscv-gnu-toolchain-20211101-36259-f4b17p/riscv-gdb/opcodes/../mkinstalldirs /opt/homebrew/Cellar/riscv-gnu-toolchain/master/share/gettext/po; \
	  else \
	    /private/tmp/riscv-gnu-toolchain-20211101-36259-f4b17p/riscv-gdb/opcodes/mkinstalldirs /opt/homebrew/Cellar/riscv-gnu-toolchain/master/share/gettext/po; \
	  fi; \
	  /usr/bin/install -c -m 644 /private/tmp/riscv-gnu-toolchain-20211101-36259-f4b17p/riscv-gdb/opcodes/po/Makefile.in.in \
			  /opt/homebrew/Cellar/riscv-gnu-toolchain/master/share/gettext/po/Makefile.in.in; \
	else \
	  : ; \
	fi
mkdir -p stamps/ && touch stamps/build-gdb-newlib

Do not report this issue to Homebrew/brew or Homebrew/core!

Traceback (most recent call last):
	27: from /opt/homebrew/Library/Homebrew/build.rb:224:in `<main>'
	26: from /opt/homebrew/Library/Homebrew/build.rb:124:in `install'
	25: from /opt/homebrew/Library/Homebrew/utils.rb:550:in `with_env'
	24: from /opt/homebrew/Library/Homebrew/build.rb:129:in `block in install'
	23: from /opt/homebrew/Library/Homebrew/formula.rb:1284:in `brew'
	22: from /opt/homebrew/Library/Homebrew/formula.rb:2442:in `stage'
	21: from /System/Library/Frameworks/Ruby.framework/Versions/2.6/usr/lib/ruby/2.6.0/forwardable.rb:230:in `stage'
	20: from /opt/homebrew/Library/Homebrew/resource.rb:96:in `stage'
	19: from /opt/homebrew/Library/Homebrew/resource.rb:121:in `unpack'
	18: from /opt/homebrew/Library/Homebrew/resource.rb:208:in `mktemp'
	17: from /opt/homebrew/Library/Homebrew/mktemp.rb:63:in `run'
	16: from /opt/homebrew/Library/Homebrew/mktemp.rb:63:in `chdir'
	15: from /opt/homebrew/Library/Homebrew/mktemp.rb:63:in `block in run'
	14: from /opt/homebrew/Library/Homebrew/resource.rb:122:in `block in unpack'
	13: from /opt/homebrew/Library/Homebrew/download_strategy.rb:102:in `stage'
	12: from /opt/homebrew/Library/Homebrew/download_strategy.rb:110:in `chdir'
	11: from /opt/homebrew/Library/Homebrew/resource.rb:126:in `block (2 levels) in unpack'
	10: from /opt/homebrew/Library/Homebrew/formula.rb:2462:in `block in stage'
	 9: from /opt/homebrew/Library/Homebrew/utils.rb:550:in `with_env'
	 8: from /opt/homebrew/Library/Homebrew/formula.rb:2463:in `block (2 levels) in stage'
	 7: from /opt/homebrew/Library/Homebrew/formula.rb:1291:in `block in brew'
	 6: from /opt/homebrew/Library/Homebrew/build.rb:134:in `block (2 levels) in install'
	 5: from /opt/homebrew/Library/Homebrew/utils.rb:550:in `with_env'
	 4: from /opt/homebrew/Library/Homebrew/build.rb:172:in `block (3 levels) in install'
	 3: from /opt/homebrew/Library/Taps/riscv/homebrew-riscv/riscv-gnu-toolchain.rb:41:in `install'
	 2: from /opt/homebrew/Library/Homebrew/formula.rb:2233:in `system'
	 1: from /opt/homebrew/Library/Homebrew/formula.rb:2233:in `open'
/opt/homebrew/Library/Homebrew/formula.rb:2297:in `block in system': Failed executing: make (BuildError)
	9: from /opt/homebrew/Library/Homebrew/brew.rb:143:in `<main>'
	8: from /opt/homebrew/Library/Homebrew/brew.rb:155:in `rescue in <main>'
	7: from /opt/homebrew/Library/Homebrew/exceptions.rb:539:in `dump'
	6: from /opt/homebrew/Library/Homebrew/exceptions.rb:485:in `issues'
	5: from /opt/homebrew/Library/Homebrew/exceptions.rb:489:in `fetch_issues'
	4: from /opt/homebrew/Library/Homebrew/utils/github.rb:61:in `issues_for_formula'
	3: from /opt/homebrew/Library/Homebrew/utils/github.rb:30:in `search_issues'
	2: from /opt/homebrew/Library/Homebrew/utils/github.rb:160:in `search'
	1: from /opt/homebrew/Library/Homebrew/utils/github/api.rb:234:in `open_rest'
/opt/homebrew/Library/Homebrew/utils/github/api.rb:304:in `raise_error': Validation Failed: [{"message"=>"The listed users and repositories cannot be searched either because the resources do not exist or you do not have permission to view them.", "resource"=>"Search", "field"=>"q", "code"=>"invalid"}] (GitHub::API::ValidationFailedError)

mnwsth, 2021-12-07:
ran into the exact same issue with 

> brew install riscv-tools

on M1 Mac, macOS Big Sur 11.4

```
/opt/homebrew/Library/Homebrew/utils/github/api.rb:304:in `raise_error': Validation Failed: [{"message"=>"The listed users and repositories cannot be searched either because the resources do not exist or you do not have permission to view them.", "resource"=>"Search", "field"=>"q", "code"=>"invalid"}] (GitHub::API::ValidationFailedError)
	from /opt/homebrew/Library/Homebrew/utils/github/api.rb:234:in `open_rest'
	from /opt/homebrew/Library/Homebrew/utils/github.rb:166:in `search'
	from /opt/homebrew/Library/Homebrew/utils/github.rb:34:in `search_issues'
	from /opt/homebrew/Library/Homebrew/utils/github.rb:67:in `issues_for_formula'
	from /opt/homebrew/Library/Homebrew/exceptions.rb:489:in `fetch_issues'
	from /opt/homebrew/Library/Homebrew/exceptions.rb:485:in `issues'
	from /opt/homebrew/Library/Homebrew/exceptions.rb:539:in `dump'
	from /opt/homebrew/Library/Homebrew/brew.rb:155:in `rescue in <main>'
	from /opt/homebrew/Library/Homebrew/brew.rb:143:in `<main>'
/opt/homebrew/Library/Homebrew/formula.rb:2306:in `block in system': Failed executing: make (BuildError)
	from /opt/homebrew/Library/Homebrew/formula.rb:2242:in `open'
	from /opt/homebrew/Library/Homebrew/formula.rb:2242:in `system'
	from /opt/homebrew/Library/Taps/riscv/homebrew-riscv/riscv-gnu-toolchain.rb:41:in `install'
	from /opt/homebrew/Library/Homebrew/build.rb:172:in `block (3 levels) in install'
	from /opt/homebrew/Library/Homebrew/utils.rb:588:in `with_env'
	from /opt/homebrew/Library/Homebrew/build.rb:134:in `block (2 levels) in install'
	from /opt/homebrew/Library/Homebrew/formula.rb:1297:in `block in brew'
	from /opt/homebrew/Library/Homebrew/formula.rb:2472:in `block (2 levels) in stage'
	from /opt/homebrew/Library/Homebrew/utils.rb:588:in `with_env'
	from /opt/homebrew/Library/Homebrew/formula.rb:2471:in `block in stage'
	from /opt/homebrew/Library/Homebrew/resource.rb:126:in `block (2 levels) in unpack'
	from /opt/homebrew/Library/Homebrew/download_strategy.rb:110:in `chdir'
	from /opt/homebrew/Library/Homebrew/download_strategy.rb:102:in `stage'
	from /opt/homebrew/Library/Homebrew/resource.rb:122:in `block in unpack'
	from /opt/homebrew/Library/Homebrew/mktemp.rb:63:in `block in run'
	from /opt/homebrew/Library/Homebrew/mktemp.rb:63:in `chdir'
	from /opt/homebrew/Library/Homebrew/mktemp.rb:63:in `run'
	from /opt/homebrew/Library/Homebrew/resource.rb:208:in `mktemp'
	from /opt/homebrew/Library/Homebrew/resource.rb:121:in `unpack'
	from /opt/homebrew/Library/Homebrew/resource.rb:96:in `stage'
	from /opt/homebrew/Library/Homebrew/vendor/portable-ruby/2.6.8/lib/ruby/2.6.0/forwardable.rb:230:in `stage'
	from /opt/homebrew/Library/Homebrew/formula.rb:2451:in `stage'
	from /opt/homebrew/Library/Homebrew/formula.rb:1290:in `brew'
	from /opt/homebrew/Library/Homebrew/build.rb:129:in `block in install'
	from /opt/homebrew/Library/Homebrew/utils.rb:588:in `with_env'
	from /opt/homebrew/Library/Homebrew/build.rb:124:in `install'
	from /opt/homebrew/Library/Homebrew/build.rb:224:in `<main>'
```

sbeamer, 2021-12-17:
I'm sorry that I can't debug these M1 issues since I don't have one. I'm going to close this and recommend folks keep adding to the existing threads: #46 #60

closed 2021-12-17 by sbeamer

cceeci99, 2021-12-21:
> ran into the exact same issue with
>
>
> on M1 Mac, macOS Big Sur 11.4
>
> ```
> /opt/homebrew/Library/Homebrew/utils/github/api.rb:304:in `raise_error': Validation Failed: [{"message"=>"The listed users and [...]

did you find solution to this I really need some help for my assigment ...

z-wenbo, 2022-01-19:
> No description provided.

Hi, have you fixed this because I've run into the exact same promblem...