riscv-software-src/homebrew-riscv #79: test installation failed

irmathebest, 2021-12-27:
Hi all. I am wondering, have anyone encountered this error when testing the installation of the riscv-gnu-toolchain? I used MacBook M1 with bigSur 11.4. Thanks. 
[image]

sbeamer, 2021-12-28:
Should be fixed now (might need to rebuild). Probably a regression from #71.

irmathebest, 2021-12-28:
> Should be fixed now (might need to rebuild). Probably a regression from #71.

Thank you so much for your help. However, I rebuild it with the command " brew reinstall --build-from-source riscv-gnu-toolchain", but the error still exists when I test it.

sreersr, 2021-12-28:
> Hi all. I am wondering, have anyone encountered this error when testing the installation of the riscv-gnu-toolchain? I used MacBook M1 with bigSur 11.4. Thanks.
> [image]

Hi I have same issue can you help me to fix this

sbeamer, 2021-12-30:
To clarify, you will need to rebuild spike (riscv-isa-sim) and riscv-pk, but you shouldn't need to rebuild riscv-gnu-toolchain.

The error was actually quite small. The compiler and emulator both worked fine. The issue was the emulator's hard coded path to check for pk was wrong, but fi you gave the full path to pk it would work. If you aren't using pk or spike, none of this would matter.

sreersr, 2022-01-01:
Hi Scott,
Thanks and I had fixed with your advise. I also need a guidance how to run a simple c code in this setup, since risk tool chain is installed using a home-brew I am not sure how to run it, please advise thanks in advance. Also do I get a RTL code for the RISCV core used here?.

sreersr, 2022-01-01:
I was trying to use RISCV toolchain with the Ariane core from https://github.com/openhwgroup/cva6 and trying to build it, but while running make Verilate the following error is observed, please advise. it will be appreciated if you give separate advise for my queries. Thanks in advance. 
[image]

sbeamer, 2022-01-02:
This repo is specifically for assisting with getting a vanilla installation of the RISC-V toolchain on macOS.

It does not provide processor RTL. For questions about a given implementation, I would follow up with the provider.

For running a simple program, you can follow the instructions from the simulator: https://github.com/riscv-software-src/riscv-isa-sim#compiling-and-running-a-simple-c-program

closed 2022-01-15 by sbeamer