riscv-software-src/homebrew-riscv #81: [Error] riscv-tools: Failed to download resource "riscv-gnu-toolchain"

PeterWrighten, 2022-01-17:
When I try installing from Homebrew, the fatal error occurres as follow:

```shell
Cloning into '/Users/apple/Library/Caches/Homebrew/riscv-gnu-toolchain--git/riscv-glibc'...
fatal: unable to connect to sourceware.org:
sourceware.org[0: 8.43.85.97]: errno=Connection refused

fatal: clone of 'git://sourceware.org/git/glibc.git' into submodule path '/Users/apple/Library/Caches/Homebrew/riscv-gnu-toolchain--git/riscv-glibc' failed
Failed to clone 'riscv-glibc'. Retry scheduled
Cloning into '/Users/apple/Library/Caches/Homebrew/riscv-gnu-toolchain--git/riscv-newlib'...
fatal: unable to connect to sourceware.org:
sourceware.org[0: 8.43.85.97]: errno=Connection refused

fatal: clone of 'git://sourceware.org/git/newlib-cygwin.git' into submodule path '/Users/apple/Library/Caches/Homebrew/riscv-gnu-toolchain--git/riscv-newlib' failed
Failed to clone 'riscv-newlib'. Retry scheduled
Cloning into '/Users/apple/Library/Caches/Homebrew/riscv-gnu-toolchain--git/riscv-glibc'...
fatal: unable to connect to sourceware.org:
sourceware.org[0: 8.43.85.97]: errno=Connection refused

fatal: clone of 'git://sourceware.org/git/glibc.git' into submodule path '/Users/apple/Library/Caches/Homebrew/riscv-gnu-toolchain--git/riscv-glibc' failed
Failed to clone 'riscv-glibc' a second time, aborting
Error: riscv-tools: Failed to download resource "riscv-gnu-toolchain"
Failure while executing; `git submodule update --init --recursive` exited with 1. Here's the output:
Submodule 'qemu' (https://git.qemu.org/git/qemu.git) registered for path 'qemu'
Submodule 'riscv-binutils' (https://github.com/riscv-collab/riscv-binutils-gdb.git) registered for path 'riscv-binutils'
Submodule 'riscv-dejagnu' (https://github.com/riscv-collab/riscv-dejagnu.git) registered for path 'riscv-dejagnu'
Submodule 'riscv-gcc' (https://github.com/riscv-collab/riscv-gcc.git) registered for path 'riscv-gcc'
Submodule 'riscv-gdb' (https://github.com/riscv-collab/riscv-binutils-gdb.git) registered for path 'riscv-gdb'
Submodule 'riscv-glibc' (git://sourceware.org/git/glibc.git) registered for path 'riscv-glibc'
Submodule 'riscv-newlib' (git://sourceware.org/git/newlib-cygwin.git) registered for path 'riscv-newlib'
Cloning into '/Users/apple/Library/Caches/Homebrew/riscv-gnu-toolchain--git/qemu'...
warning: redirecting to https://gitlab.com/qemu-project/qemu.git/
Cloning into '/Users/apple/Library/Caches/Homebrew/riscv-gnu-toolchain--git/riscv-binutils'...
Cloning into '/Users/apple/Library/Caches/Homebrew/riscv-gnu-toolchain--git/riscv-dejagnu'...
Cloning into '/Users/apple/Library/Caches/Homebrew/riscv-gnu-toolchain--git/riscv-gcc'...
Cloning into '/Users/apple/Library/Caches/Homebrew/riscv-gnu-toolchain--git/riscv-gdb'...
Cloning into '/Users/apple/Library/Caches/Homebrew/riscv-gnu-toolchain--git/riscv-glibc'...
fatal: unable to connect to sourceware.org:
sourceware.org[0: 8.43.85.97]: errno=Connection refused

fatal: clone of 'git://sourceware.org/git/glibc.git' into submodule path '/Users/apple/Library/Caches/Homebrew/riscv-gnu-toolchain--git/riscv-glibc' failed
Failed to clone 'riscv-glibc'. Retry scheduled
Cloning into '/Users/apple/Library/Caches/Homebrew/riscv-gnu-toolchain--git/riscv-newlib'...
fatal: unable to connect to sourceware.org:
sourceware.org[0: 8.43.85.97]: errno=Connection refused

fatal: clone of 'git://sourceware.org/git/newlib-cygwin.git' into submodule path '/Users/apple/Library/Caches/Homebrew/riscv-gnu-toolchain--git/riscv-newlib' failed
Failed to clone 'riscv-newlib'. Retry scheduled
Cloning into '/Users/apple/Library/Caches/Homebrew/riscv-gnu-toolchain--git/riscv-glibc'...
fatal: unable to connect to sourceware.org:
sourceware.org[0: 8.43.85.97]: errno=Connection refused

fatal: clone of 'git://sourceware.org/git/glibc.git' into submodule path '/Users/apple/Library/Caches/Homebrew/riscv-gnu-toolchain--git/riscv-glibc' failed
Failed to clone 'riscv-glibc' a second time, aborting
```

sbeamer, 2022-01-17:
If we are lucky, this might be a temporary connectivity issue? Please retry and see if it persists. I would also check if you can clone any of those repos directly.

PeterWrighten, 2022-01-17:
I checked it a few minutes ago, and it worked out right now. Yeah, it seems to be exactly a temporary connectivity issue. Thanks for your patient reply :)

closed 2022-01-17 by PeterWrighten