riscv-software-src/homebrew-riscv #86: install failed on macos

zhenjunMa, 2022-05-01:
# System Info
```
Darwin Kernel Version 19.5.0: Tue May 26 20:41:44 PDT 2020; root:xnu-6153.121.2~2/RELEASE_X86_64 x86_64
```

# Command
```
brew tap riscv-software-src/riscv
brew install riscv-tools
```

# Error
```
==> Cloning https://github.com/riscv/riscv-gnu-toolchain.git
Updating /Users/xxx/Library/Caches/Homebrew/riscv-gnu-toolchain--git
==> Checking out branch master
Already on 'master'
Your branch is up to date with 'origin/master'.
HEAD is now at 9249802 Merge pull request #1062 from palmer-dabbelt/libsanitizer
Entering 'glibc'
Entering 'musl'
Entering 'newlib'
Entering 'qemu'
Entering 'riscv-binutils'
Entering 'riscv-dejagnu'
Entering 'riscv-gcc'
Entering 'riscv-gdb'
fatal: Needed a single revision
Unable to find current revision in submodule path 'newlib'
Error: riscv-tools: Failed to download resource "riscv-gnu-toolchain"
Failure while executing; `git submodule update --init --recursive` exited with 1. Here's the output:
fatal: Needed a single revision
Unable to find current revision in submodule path 'newlib'
```

How could i solve this problem?  Any suggestions?

chenqwwq, 2022-06-26:
我也遇到了。。

zhenjunMa, 2022-06-27:
@chenqwwq 
You can try to solve this problem according to the steps in this document: https://zhayujie.com/mit6828-env.html

chenqwwq, 2022-06-27:
@zhenjunMa thx

closed 2022-07-17 by sbeamer