riscv-software-src/homebrew-riscv #87: fatal: Unable to find current revision in submodule path 'glibc'

MrZLeo, 2022-05-27:
I got error said:

```
fatal: Unable to find current revision in submodule path 'glibc'
Error: riscv-tools: Failed to download resource "riscv-gnu-toolchain"
Failure while executing; `git submodule update --init --recursive` exited with 128. Here's the output:
fatal: Unable to find current revision in submodule path 'glibc'
```

I don't know how to fix it. Anyone can help?

Sped0n, 2022-07-16:
you can try to clone riscv-gnu-toolchain directly from GitHub, then copy all the files in the folder to where your brew cache is, like```/Users/spedon/Library/Caches/Homebrew/riscv-gnu-toolchain--git```.

sbeamer, 2022-07-17:
The recent change (3a5cb0c811f7b7312a243fb27675a53264db2b89) to reduce the number of submodules pulled in may help with this.

closed 2022-07-22 by MrZLeo