riscv-software-src/homebrew-riscv #89: Homebrew installation error

japrozs, 2022-08-21:
I ran the following command and was facing some issues with installing the toolchain on my Mac
```
$ brew tap riscv-software-src/riscv
$ brew install riscv-tools
Warning: No available formula with the name "flock" (dependency of riscv-software-src/riscv/riscv-tools). Did you mean loc or flow?
==> Searching for similarly named formulae...
These similarly named formulae were found:
loc                                                       flow
To install one of them, run (for example):
  brew install loc
Found a cask named "flock" instead. Try
  brew install --cask flock

```
i'm running macOS `10.15.7`
my machine
```
Darwin Pritpals-MacBook-Pro.local 19.6.0 Darwin Kernel Version 19.6.0: Mon Apr 18 21:50:40 PDT 2022; root:xnu-6153.141.62~1/RELEASE_X86_64 x86_64
```

homebrew
```
Homebrew 3.4.0
Homebrew/homebrew-core (git revision 4e9981d0db5; last commit 2022-03-04)
Homebrew/homebrew-cask (git revision fe09f6227a; last commit 2022-03-04)
```

sbeamer, 2022-11-04:
I'm confused by this error since `flock` still seems to be available in Homebrew?

japrozs, 2022-11-05:
i tried installing it a few days ago, and there's seems to be a confusion over which `flock` to install as their seem to be more than one package on `homebrew` referenced by the name `flock`

japrozs, 2022-11-05:
> I'm confused by this error since `flock` still seems to be available in Homebrew?

does this method of installation work on your machine?

sbeamer, 2022-11-05:
In Homebrew's core, formula names should be unique: https://github.com/Homebrew/homebrew-core/blob/HEAD/Formula/flock.rb

If you are seeing two copies of flock, perhaps you have a tap with that name? The original error from above shows 0 flock formula found.

closed 2022-12-17 by sbeamer