riscv-software-src/homebrew-riscv #92: Homebrew Installation Error

ThePerfectComputer, 2022-08-23:
Running ``brew install riscv-tools`` on macOS Monterey Apple Silicon fails on:

```
==> git submodule update --init --recursive riscv-binutils
Last 15 lines from /Users/redacted/Library/Logs/Homebrew/riscv-gnu-toolchain/02.git:
2022-08-23 22:34:51 +0000

git
submodule
update
--init
--recursive
riscv-binutils

error: pathspec 'riscv-binutils' did not match any file(s) known to git

Do not report this issue to Homebrew/brew or Homebrew/core!

These open issues may also help:
Sync the binutils name with riscv-gnu-toolchain repo https://github.com/riscv-software-src/homebrew-riscv/pull/91
```

brandaogbs, 2022-08-28:
While we don't have the official fix, you can try the solution I've reported on [riscv-collab/riscv-gnu-toolchain/issues/1117#issuecomment-1229446707](riscv-collab/riscv-gnu-toolchain/issues/1117#issuecomment-1229446707)

dianhsu, 2022-08-28:
> While we don't have the official fix, you can try the solution I've reported on [riscv-collab/riscv-gnu-toolchain/issues/1117#issuecomment-1229446707](https://github.com/riscv-collab/riscv-gnu-toolchain/issues/1117#issuecomment-1229446707)

Thanks, it works.