riscv-software-src/homebrew-riscv #93: Homebrew test Warning

Healthcliff-Ding, 2022-10-20:
It seems my installation directly via `brew` didn't pass the test.
```bash
% brew test riscv-tools
==> Testing riscv-software-src/riscv/riscv-tools
==> /opt/homebrew/opt/riscv-gnu-toolchain/bin/riscv64-unknown-elf-gcc /private/tmp/riscv-tools-test-20221020-11032-1bvorer/hello.c
==> spike pk /private/tmp/riscv-tools-test-20221020-10771-42v28m/a.out
<stdin>:48.24-56.7: Warning (interrupt_provider): /soc/plic@c000000: Missing #address-cells in interrupt provider
```
My machine is equipped with M1 silicon, and OS is Monterey 12.6.

Appreciate for your notice, and have a nice day.

Healthcliff-Ding, 2022-10-20:
I've found a similar [issue](https://github.com/riscv-software-src/riscv-isa-sim/issues/784).

sbeamer, 2022-11-05:
I can't reproduce this issue. The PR you linked to should have been resolved by a fix a year ago. If it helps any, I did just push new bottles.

closed 2022-12-17 by sbeamer