riscv-software-src/homebrew-riscv #95: error when installing dependency:riscv-isa-sim

SeddonShen, 2022-11-18:
HEAD is now at 425e8583 Merge branch 'plctlab-plct-zce-upstream'
==> Cloning https://github.com/riscv/riscv-pk.git
Updating /Users/seddonshen/Library/Caches/Homebrew/riscv-pk--git
==> Checking out branch master
Already on 'master'
Your branch is up to date with 'origin/master'.
HEAD is now at 7e9b671 Specify text section in rest_of_boot_loader definition
==> Downloading https://github.com/riscv/riscv-tools/archive/homebrew.tar.gz
Already downloaded: /Users/seddonshen/Library/Caches/Homebrew/downloads/5e080ac67877162e34a9e6701d88df637ec2f4abb27be9857918a142c6a3477e--riscv-tools-homebrew.tar.gz
==> Installing riscv-tools from riscv-software-src/riscv
==> Installing dependencies for riscv-software-src/riscv/riscv-tools: riscv-isa-sim and riscv-pk
==> Installing riscv-software-src/riscv/riscv-tools dependency: riscv-isa-sim
==> ../configure --prefix=/opt/homebrew/Cellar/riscv-isa-sim/main --with-target=../../riscv-pk/main/riscv64-unknown-elf --without-boost 
==> make
Last 15 lines from /Users/seddonshen/Library/Logs/Homebrew/riscv-isa-sim/02.make:
../riscv/decode.h:167:10: error: unknown type name 'uint'; did you mean 'int'?
    for (uint i = 5; i <= rlist; i++)
         ^~~~
         int
clang++ -MMD -MP  -DPREFIX=\"/opt/homebrew/Cellar/riscv-isa-sim/main\" -Wall -Wno-unused -Wno-nonportable-include-path -g -O2 -fPIC -g -O2 -DPREFIX=\"/opt/homebrew/Cellar/riscv-isa-sim/main\" -Wall -Wno-unused -Wno-nonportable-include-path -g -O2 -fPIC -std=c++17 -g -O2   -I. -I.. -I../fesvr -I../riscv -I../disasm -I../customext -I../fdt -I../softfloat -I../spike_main -I../spike_dasm     -c ../fesvr/dummy.cc
clang++ -MMD -MP  -DPREFIX=\"/opt/homebrew/Cellar/riscv-isa-sim/main\" -Wall -Wno-unused -Wno-nonportable-include-path -g -O2 -fPIC -g -O2 -DPREFIX=\"/opt/homebrew/Cellar/riscv-isa-sim/main\" -Wall -Wno-unused -Wno-nonportable-include-path -g -O2 -fPIC -std=c++17 -g -O2   -I. -I.. -I../fesvr -I../riscv -I../disasm -I../customext -I../fdt -I../softfloat -I../spike_main -I../spike_dasm     -c ../fesvr/option_parser.cc
clang++ -MMD -MP  -DPREFIX=\"/opt/homebrew/Cellar/riscv-isa-sim/main\" -Wall -Wno-unused -Wno-nonportable-include-path -g -O2 -fPIC -g -O2 -DPREFIX=\"/opt/homebrew/Cellar/riscv-isa-sim/main\" -Wall -Wno-unused -Wno-nonportable-include-path -g -O2 -fPIC -std=c++17 -g -O2   -I. -I.. -I../fesvr -I../riscv -I../disasm -I../customext -I../fdt -I../softfloat -I../spike_main -I../spike_dasm     -c ../fesvr/term.cc
clang++ -MMD -MP  -DPREFIX=\"/opt/homebrew/Cellar/riscv-isa-sim/main\" -Wall -Wno-unused -Wno-nonportable-include-path -g -O2 -fPIC -g -O2 -DPREFIX=\"/opt/homebrew/Cellar/riscv-isa-sim/main\" -Wall -Wno-unused -Wno-nonportable-include-path -g -O2 -fPIC -std=c++17 -g -O2   -I. -I.. -I../fesvr -I../riscv -I../disasm -I../customext -I../fdt -I../softfloat -I../spike_main -I../spike_dasm     -c ../fesvr/tsi.cc
for insn in amoadd_d amoadd_w amoand_d amoand_w amomax_d amomaxu_d amomaxu_w amomax_w amomin_d amominu_d amominu_w amomin_w amoor_d amoor_w amoswap_d amoswap_w amoxor_d amoxor_w lr_d lr_w sc_d sc_w c_add c_addi c_addi4spn c_addw c_and c_andi c_beqz c_bnez c_ebreak c_fld c_fldsp c_flw c_flwsp c_fsd c_fsdsp c_fsw c_fswsp c_j c_jal c_jalr c_jr c_li c_lui c_lw c_lwsp c_mv c_or c_slli c_srai c_srli c_sub c_subw c_sw c_swsp c_xor add addi addiw addw and andi auipc beq bge bgeu blt bltu bne jal jalr lb lbu ld lh lhu lui lw lwu or ori sb sd sh sll slli slliw sllw slt slti sltiu sltu sra srai sraiw sraw srl srli srliw srlw sub subw sw xor xori fence fence_i div divu divuw divw mul mulh mulhsu mulhu mulw rem remu remuw remw fadd_s fclass_s fcvt_l_s fcvt_lu_s fcvt_s_l fcvt_s_lu fcvt_s_w fcvt_s_wu fcvt_w_s fcvt_wu_s fdiv_s feq_s fle_s flt_s flw fmadd_s fmax_s fmin_s fmsub_s fmul_s fmv_w_x fmv_x_w fnmadd_s fnmsub_s fsgnj_s fsgnjn_s fsgnjx_s fsqrt_s fsub_s fsw fadd_d fclass_d fcvt_d_l fcvt_d_lu fcvt_d_q fcvt_d_s fcvt_d_w fcvt_d_wu fcvt_l_d fcvt_lu_d fcvt_s_d fcvt_w_d fcvt_wu_d fdiv_d feq_d fld fle_d flt_d fmadd_d fmax_d fmin_d fmsub_d fmul_d fmv_d_x fmv_x_d fnmadd_d fnmsub_d fsd fsgnj_d fsgnjn_d fsgnjx_d fsqrt_d fsub_d fadd_h fclass_h fcvt_l_h fcvt_lu_h fcvt_d_h fcvt_h_d fcvt_h_l fcvt_h_lu fcvt_h_q fcvt_h_s fcvt_h_w fcvt_h_wu fcvt_q_h fcvt_s_h fcvt_w_h fcvt_wu_h fdiv_h feq_h fle_h flh flt_h fmadd_h fmax_h fmin_h fmsub_h fmul_h fmv_h_x fmv_x_h fnmadd_h fnmsub_h fsgnj_h fsgnjn_h fsgnjx_h fsh fsqrt_h fsub_h fadd_q fclass_q fcvt_l_q fcvt_lu_q fcvt_q_d fcvt_q_l fcvt_q_lu fcvt_q_s fcvt_q_w fcvt_q_wu fcvt_s_q fcvt_w_q fcvt_wu_q fdiv_q feq_q fle_q flq flt_q fmadd_q fmax_q fmin_q fmsub_q fmul_q fnmadd_q fnmsub_q fsgnj_q fsgnjn_q fsgnjx_q fsq fsqrt_q fsub_q add_uw andn bdecompress bdecompressw bcompress bcompressw bfp bfpw bmatflip bmator bmatxor sh1add sh1add_uw sh2add sh2add_uw sh3add sh3add_uw clmul clmulh clmulr clz clzw cmix cmov crc32_b crc32c_b crc32c_d crc32c_h crc32c_w crc32_d crc32_h crc32_w ctz ctzw fsl fslw fsr fsri fsriw fsrw gorc gorci gorciw gorcw grev grevi greviw grevw max maxu min minu orn pack packh packu packuw packw cpop cpopw rol rolw ror rori roriw rorw bclr bclri bext bexti binv binvi bset bseti sext_b sext_h shfl shfli shflw slli_uw slo sloi sloiw slow sro sroi sroiw srow unshfl unshfli unshflw xnor xperm4 xperm8 xperm16 xperm32 aes32dsi aes32dsmi aes32esi aes32esmi aes64ds aes64dsm aes64es aes64esm aes64ks1i aes64ks2 aes64im sha256sig0 sha256sig1 sha256sum0 sha256sum1 sha512sig0 sha512sig0h sha512sig0l sha512sig1 sha512sig1h sha512sig1l sha512sum0 sha512sum0r sha512sum1 sha512sum1r sm3p0 sm3p1 sm4ed sm4ks vfadd_vf vfadd_vv vfclass_v vfcvt_f_x_v vfcvt_f_xu_v vfcvt_rtz_x_f_v vfcvt_rtz_xu_f_v vfcvt_x_f_v vfcvt_xu_f_v vfdiv_vf vfdiv_vv vfmacc_vf vfmacc_vv vfmadd_vf vfmadd_vv vfmax_vf vfmax_vv vfmerge_vfm vfmin_vf vfmin_vv vfmsac_vf vfmsac_vv vfmsub_vf vfmsub_vv vfmul_vf vfmul_vv vfmv_f_s vfmv_s_f vfmv_v_f vfncvt_f_f_w vfncvt_f_x_w vfncvt_f_xu_w vfncvt_rod_f_f_w vfncvt_rtz_x_f_w vfncvt_rtz_xu_f_w vfncvt_x_f_w vfncvt_xu_f_w vfnmacc_vf vfnmacc_vv vfnmadd_vf vfnmadd_vv vfnmsac_vf vfnmsac_vv vfnmsub_vf vfnmsub_vv vfrdiv_vf vfredmax_vs vfredmin_vs vfredosum_vs vfredusum_vs vfrec7_v vfrsub_vf vfrsqrt7_v vfsgnj_vf vfsgnj_vv vfsgnjn_vf vfsgnjn_vv vfsgnjx_vf vfsgnjx_vv vfsqrt_v vfslide1down_vf vfslide1up_vf vfsub_vf vfsub_vv vfwadd_vf vfwadd_vv vfwadd_wf vfwadd_wv vfwcvt_f_f_v vfwcvt_f_x_v vfwcvt_f_xu_v vfwcvt_rtz_x_f_v vfwcvt_rtz_xu_f_v vfwcvt_x_f_v vfwcvt_xu_f_v vfwmacc_vf vfwmacc_vv vfwmsac_vf vfwmsac_vv vfwmul_vf vfwmul_vv vfwnmacc_vf vfwnmacc_vv vfwnmsac_vf vfwnmsac_vv vfwredosum_vs vfwredusum_vs vfwsub_vf vfwsub_vv vfwsub_wf vfwsub_wv vmfeq_vf vmfeq_vv vmfge_vf vmfgt_vf vmfle_vf vmfle_vv vmflt_vf vmflt_vv vmfne_vf vmfne_vv vaadd_vv vaaddu_vv vaadd_vx vaaddu_vx vadc_vim vadc_vvm vadc_vxm vadd_vi vadd_vv vadd_vx vand_vi vand_vv vand_vx vasub_vv vasubu_vv vasub_vx vasubu_vx vcompress_vm vcpop_m vdiv_vv vdiv_vx vdivu_vv vdivu_vx vid_v viota_m vmacc_vv vmacc_vx vmadc_vv vmadc_vx vmadc_vi vmadc_vim vmadc_vvm vmadc_vxm vmadd_vv vmadd_vx vmand_mm vmandn_mm vmax_vv vmax_vx vmaxu_vv vmaxu_vx vmerge_vim vmerge_vvm vmerge_vxm vfirst_m vmin_vv vmin_vx vminu_vv vminu_vx vmnand_mm vmnor_mm vmor_mm vmorn_mm vmsbc_vv vmsbc_vx vmsbc_vvm vmsbc_vxm vmsbf_m vmseq_vi vmseq_vv vmseq_vx vmsgt_vi vmsgt_vx vmsgtu_vi vmsgtu_vx vmsif_m vmsle_vi vmsle_vv vmsle_vx vmsleu_vi vmsleu_vv vmsleu_vx vmslt_vv vmslt_vx vmsltu_vv vmsltu_vx vmsne_vi vmsne_vv vmsne_vx vmsof_m vmul_vv vmul_vx vmulh_vv vmulh_vx vmulhsu_vv vmulhsu_vx vmulhu_vv vmulhu_vx vmv_s_x vmv_v_i vmv_v_v vmv_v_x vmv_x_s vmv1r_v vmv2r_v vmv4r_v vmv8r_v vmxnor_mm vmxor_mm vnclip_wi vnclip_wv vnclip_wx vnclipu_wi vnclipu_wv vnclipu_wx vnmsac_vv vnmsac_vx vnmsub_vv vnmsub_vx vnsra_wi vnsra_wv vnsra_wx vnsrl_wi vnsrl_wv vnsrl_wx vor_vi vor_vv vor_vx vredand_vs vredmax_vs vredmaxu_vs vredmin_vs vredminu_vs vredor_vs vredsum_vs vredxor_vs vrem_vv vrem_vx vremu_vv vremu_vx vrgather_vi vrgather_vv vrgather_vx vrgatherei16_vv vrsub_vi vrsub_vx vsadd_vi vsadd_vv vsadd_vx vsaddu_vi vsaddu_vv vsaddu_vx vsbc_vvm vsbc_vxm vsext_vf2 vsext_vf4 vsext_vf8 vslide1down_vx vslide1up_vx vslidedown_vi vslidedown_vx vslideup_vi vslideup_vx vsll_vi vsll_vv vsll_vx vsmul_vv vsmul_vx vsra_vi vsra_vv vsra_vx vsrl_vi vsrl_vv vsrl_vx vssra_vi vssra_vv vssra_vx vssrl_vi vssrl_vv vssrl_vx vssub_vv vssub_vx vssubu_vv vssubu_vx vsub_vv vsub_vx vwadd_vv vwadd_vx vwadd_wv vwadd_wx vwaddu_vv vwaddu_vx vwaddu_wv vwaddu_wx vwmacc_vv vwmacc_vx vwmaccsu_vv vwmaccsu_vx vwmaccu_vv vwmaccu_vx vwmaccus_vx vwmul_vv vwmul_vx vwmulsu_vv vwmulsu_vx vwmulu_vv vwmu [... cut ...] kmsxda32 smds32 smdrs32 smxds32 sraiw_u pkbt32 pktb32 clz32 csrrc csrrci csrrs csrrsi csrrw csrrwi dret ebreak ecall mret sfence_vma sret wfi sfence_w_inval sfence_inval_ir sinval_vma hinval_vvma hinval_gvma cbo_clean cbo_flush cbo_inval cbo_zero ; do \
		printf 'DEFINE_INSN(%s)\n' "${insn}" ; \
	done > insn_list.h.tmp
1 error generated.
make: *** [htif.o] Error 1
make: *** Waiting for unfinished jobs....
mv insn_list.h.tmp insn_list.h

Do not report this issue to Homebrew/brew or Homebrew/core!

SeddonShen, 2022-11-18:
I don't why occur this compile problem, i have new installed macos 13.0.1

tdudz, 2022-11-19:
i have the same issue, MBA M1 on macOS 13.0.1

chenjunr6745, 2022-11-20:
i have the same issue too, MBP M1pro on macOS 12.0.1

sbeamer, 2022-12-21:
I have been unable to reproduce this, both on macOS 12 and now macOS 13 (after upgrading). The error logs above are truncated, but there are two things to consider:

- It appears to be using clang++ explicitly. Although g++ is aliased to clang++ on the macs, it makes me wonder if something was changed about the build configuration.
- Errors like this have occurred over the years, and have sometimes been a result of a change to https://github.com/riscv-software-src/riscv-isa-sim itself that isn't clang compatible. Sometimes something as small as adding `include <string>` can fix the affected files.

sbeamer, 2023-03-31:
Closing for inactivity. Please reopen if new content.

closed 2023-03-31 by sbeamer