riscv-software-src/homebrew-riscv #97: Would it be possible to create precompiled binaries for macOS Ventura, both Intel and Apple Silicon?

ddribin, 2023-03-19:
The `README` states:

> If you have macOS Monterey (12), `riscv-tools` will be installed from precompiled binaries. If you do not have Monterey (for x86), `riscv-tools` will be built from source.

[macOS Ventura (13)][ventura] has been out since October 2022 (about 5 months, at the time of filing). Would it be possible to create precompiled  binaries for Ventura? Ideally both Intel and Apple Silicon.

Thanks!

[ventura]: https://en.wikipedia.org/wiki/MacOS_Ventura

sbeamer, 2023-03-31:
MacOS Ventura ✅ (just pushed new bottles)

I still don't have Apple Silicon, but I would be happy to accept builds. If someone could run the following and send me both the text output and the files produced, we could include them. A PR is even better:

```
brew remove --ignore-dependencies riscv-isa-sim
brew install --build-bottle riscv-isa-sim
brew bottle riscv-isa-sim

brew remove --ignore-dependencies riscv-gnu-toolchain
brew install --build-bottle riscv-gnu-toolchain
brew bottle riscv-gnu-toolchain

brew remove --ignore-dependencies riscv-pk
brew install --build-bottle riscv-pk
brew bottle riscv-pk
```

closed 2023-03-31 by sbeamer

ddribin, 2023-04-02:
Thank you! My Apple Silicon machine is not yet running Ventura (only my Intel machine is). Once it is, I could give this a shot. I've never built a bottle before. Is it okay to run those commands in an existing Homebrew install? Or should it be done in a "clean" Homebrew install?