riscv-software-src/homebrew-riscv #99: cannot install on mac m1

MusLead, 2023-04-20:
I have seen some of you who solve this problem with removing directory `riscv-gnu-toolchain--git` in Cache [#73](https://github.com/riscv-software-src/homebrew-riscv/issues/73) But it does not work well. I still get a problem like this. I would love to see a solution from you guys

brew install riscv-tools
==> Fetching dependencies for riscv/riscv/riscv-tools: riscv-pk
==> Fetching riscv/riscv/riscv-pk
==> Cloning https://github.com/riscv/riscv-pk.git
Updating /Users/aslam/Library/Caches/Homebrew/riscv-pk--git
==> Checking out branch master
Already on 'master'
Your branch is up to date with 'origin/master'.
HEAD is now at 3ed18cf Revert "Revert "SBI emulation of reads and writes to perf counters and config (#98)""
==> Fetching riscv/riscv/riscv-tools
==> Downloading https://github.com/riscv/riscv-tools/archive/homebrew.tar.gz
Already downloaded: /Users/aslam/Library/Caches/Homebrew/downloads/5e080ac67877162e34a9e6701d88df637ec2f4abb27be9857918a142c6a3477e--riscv-tools-homebrew.tar.gz
==> Installing riscv-tools from riscv/riscv
==> Installing dependencies for riscv/riscv/riscv-tools: riscv-pk
==> Installing riscv/riscv/riscv-tools dependency: riscv-pk
==> ../configure --prefix=/opt/homebrew/Cellar/riscv-pk/main --host=riscv64-unknown-elf
==> make install
Last 15 lines from /Users/aslam/Library/Logs/Homebrew/riscv-pk/02.make:
riscv64-unknown-elf-gcc -MMD -MP -Wall -Werror -D__NO_INLINE__ -mcmodel=medany -O2 -std=gnu99 -Wno-unused -Wno-attributes -fno-delete-null-pointer-checks -fno-PIE    -DBBL_LOGO_FILE=\"bbl_logo_file\" -DMEM_START=0x80000000 -fno-stack-protector -U_FORTIFY_SOURCE -DBBL_PAYLOAD=\"bbl_payload\" -I. -I../pk -I../bbl -I../softfloat -I../dummy_payload -I../machine -I../util -c ../machine/uart.c
riscv64-unknown-elf-gcc -MMD -MP -Wall -Werror -D__NO_INLINE__ -mcmodel=medany -O2 -std=gnu99 -Wno-unused -Wno-attributes -fno-delete-null-pointer-checks -fno-PIE    -DBBL_LOGO_FILE=\"bbl_logo_file\" -DMEM_START=0x80000000 -fno-stack-protector -U_FORTIFY_SOURCE -DBBL_PAYLOAD=\"bbl_payload\" -I. -I../pk -I../bbl -I../softfloat -I../dummy_payload -I../machine -I../util -c ../machine/uart16550.c
riscv64-unknown-elf-gcc -MMD -MP -Wall -Werror -D__NO_INLINE__ -mcmodel=medany -O2 -std=gnu99 -Wno-unused -Wno-attributes -fno-delete-null-pointer-checks -fno-PIE    -DBBL_LOGO_FILE=\"bbl_logo_file\" -DMEM_START=0x80000000 -fno-stack-protector -U_FORTIFY_SOURCE -DBBL_PAYLOAD=\"bbl_payload\" -I. -I../pk -I../bbl -I../softfloat -I../dummy_payload -I../machine -I../util -c ../machine/uart_litex.c
riscv64-unknown-elf-gcc -MMD -MP -Wall -Werror -D__NO_INLINE__ -mcmodel=medany -O2 -std=gnu99 -Wno-unused -Wno-attributes -fno-delete-null-pointer-checks -fno-PIE    -DBBL_LOGO_FILE=\"bbl_logo_file\" -DMEM_START=0x80000000 -fno-stack-protector -U_FORTIFY_SOURCE -DBBL_PAYLOAD=\"bbl_payload\" -I. -I../pk -I../bbl -I../softfloat -I../dummy_payload -I../machine -I../util -c ../machine/finisher.c
riscv64-unknown-elf-gcc -MMD -MP -Wall -Werror -D__NO_INLINE__ -mcmodel=medany -O2 -std=gnu99 -Wno-unused -Wno-attributes -fno-delete-null-pointer-checks -fno-PIE    -DBBL_LOGO_FILE=\"bbl_logo_file\" -DMEM_START=0x80000000 -fno-stack-protector -U_FORTIFY_SOURCE -DBBL_PAYLOAD=\"bbl_payload\" -I. -I../pk -I../bbl -I../softfloat -I../dummy_payload -I../machine -I../util -c ../machine/misaligned_ldst.c
riscv64-unknown-elf-gcc -MMD -MP -Wall -Werror -D__NO_INLINE__ -mcmodel=medany -O2 -std=gnu99 -Wno-unused -Wno-attributes -fno-delete-null-pointer-checks -fno-PIE    -DBBL_LOGO_FILE=\"bbl_logo_file\" -DMEM_START=0x80000000 -fno-stack-protector -U_FORTIFY_SOURCE -DBBL_PAYLOAD=\"bbl_payload\" -I. -I../pk -I../bbl -I../softfloat -I../dummy_payload -I../machine -I../util -c ../machine/flush_icache.c
riscv64-unknown-elf-gcc -MMD -MP -Wall -Werror -D__NO_INLINE__ -mcmodel=medany -O2 -std=gnu99 -Wno-unused -Wno-attributes -fno-delete-null-pointer-checks -fno-PIE    -DBBL_LOGO_FILE=\"bbl_logo_file\" -DMEM_START=0x80000000 -fno-stack-protector -U_FORTIFY_SOURCE -DBBL_PAYLOAD=\"bbl_payload\" -I. -I../pk -I../bbl -I../softfloat -I../dummy_payload -I../machine -I../util -c ../machine/mentry.S
../machine/flush_icache.c: Assembler messages:
../machine/flush_icache.c:4: Error: unrecognized opcode `fence.i', extension `zifencei' required
riscv64-unknown-elf-gcc -MMD -MP -Wall -Werror -D__NO_INLINE__ -mcmodel=medany -O2 -std=gnu99 -Wno-unused -Wno-attributes -fno-delete-null-pointer-checks -fno-PIE    -DBBL_LOGO_FILE=\"bbl_logo_file\" -DMEM_START=0x80000000 -fno-stack-protector -U_FORTIFY_SOURCE -DBBL_PAYLOAD=\"bbl_payload\" -I. -I../pk -I../bbl -I../softfloat -I../dummy_payload -I../machine -I../util -c ../machine/fp_asm.S
make: *** [flush_icache.o] Error 1
make: *** Waiting for unfinished jobs....
../machine/mentry.S: Assembler messages:
../machine/mentry.S:93: Error: unrecognized opcode `fence.i', extension `zifencei' required
make: *** [mentry.o] Error 1

If reporting this issue please do so at (not Homebrew/brew or Homebrew/homebrew-core):
  https://github.com/riscv/homebrew-riscv/issues

/opt/homebrew/Library/Homebrew/utils/github/api.rb:315:in `raise_error': Validation Failed: [{"message"=>"The listed users and repositories cannot be searched either because the resources do not exist or you do not have permission to view them.", "resource"=>"Search", "field"=>"q", "code"=>"invalid"}] (GitHub::API::ValidationFailedError)
	from /opt/homebrew/Library/Homebrew/utils/github/api.rb:241:in `open_rest'
	from /opt/homebrew/Library/Homebrew/utils/github.rb:180:in `search'
	from /opt/homebrew/Library/Homebrew/utils/github.rb:184:in `search_results_items'
	from /opt/homebrew/Library/Homebrew/utils/github.rb:38:in `search_issues'
	from /opt/homebrew/Library/Homebrew/utils/github.rb:64:in `issues_for_formula'
	from /opt/homebrew/Library/Homebrew/exceptions.rb:498:in `fetch_issues'
	from /opt/homebrew/Library/Homebrew/exceptions.rb:493:in `issues'
	from /opt/homebrew/Library/Homebrew/exceptions.rb:549:in `dump'
	from /opt/homebrew/Library/Homebrew/brew.rb:151:in `rescue in <main>'
	from /opt/homebrew/Library/Homebrew/brew.rb:139:in `<main>'
/opt/homebrew/Library/Homebrew/formula.rb:2551:in `block in system': Failed executing: make install (BuildError)
	from /opt/homebrew/Library/Homebrew/formula.rb:2487:in `open'
	from /opt/homebrew/Library/Homebrew/formula.rb:2487:in `system'
	from /opt/homebrew/Library/Taps/riscv/homebrew-riscv/riscv-pk.rb:26:in `block in install'
	from /System/Library/Frameworks/Ruby.framework/Versions/2.6/usr/lib/ruby/2.6.0/fileutils.rb:128:in `chdir'
	from /System/Library/Frameworks/Ruby.framework/Versions/2.6/usr/lib/ruby/2.6.0/fileutils.rb:128:in `cd'
	from /opt/homebrew/Library/Taps/riscv/homebrew-riscv/riscv-pk.rb:22:in `install'
	from /opt/homebrew/Library/Homebrew/build.rb:177:in `block (3 levels) in install'
	from /opt/homebrew/Library/Homebrew/extend/kernel.rb:500:in `with_env'
	from /opt/homebrew/Library/Homebrew/build.rb:139:in `block (2 levels) in install'
	from /opt/homebrew/Library/Homebrew/formula.rb:1327:in `block in brew'
	from /opt/homebrew/Library/Homebrew/formula.rb:2718:in `block (2 levels) in stage'
	from /opt/homebrew/Library/Homebrew/extend/kernel.rb:500:in `with_env'
	from /opt/homebrew/Library/Homebrew/formula.rb:2717:in `block in stage'
	from /opt/homebrew/Library/Homebrew/resource.rb:159:in `block (2 levels) in unpack'
	from /opt/homebrew/Library/Homebrew/download_strategy.rb:111:in `chdir'
	from /opt/homebrew/Library/Homebrew/download_strategy.rb:103:in `stage'
	from /opt/homebrew/Library/Homebrew/resource.rb:155:in `block in unpack'
	from /opt/homebrew/Library/Homebrew/mktemp.rb:77:in `block in run'
	from /opt/homebrew/Library/Homebrew/mktemp.rb:77:in `chdir'
	from /opt/homebrew/Library/Homebrew/mktemp.rb:77:in `run'
	from /opt/homebrew/Library/Homebrew/resource.rb:267:in `stage_resource'
	from /opt/homebrew/Library/Homebrew/resource.rb:154:in `unpack'
	from /opt/homebrew/Library/Homebrew/resource.rb:128:in `stage'
	from /System/Library/Frameworks/Ruby.framework/Versions/2.6/usr/lib/ruby/2.6.0/forwardable.rb:230:in `stage'
	from /opt/homebrew/Library/Homebrew/formula.rb:2697:in `stage'
	from /opt/homebrew/Library/Homebrew/formula.rb:1320:in `brew'
	from /opt/homebrew/Library/Homebrew/build.rb:133:in `block in install'
	from /opt/homebrew/Library/Homebrew/extend/kernel.rb:500:in `with_env'
	from /opt/homebrew/Library/Homebrew/build.rb:128:in `install'
	from /opt/homebrew/Library/Homebrew/build.rb:229:in `<main>'

wstevens2090, 2023-04-26:
Also having an issue with the 'zifencei' required for a M1 Pro MacBook.

rain7996, 2023-04-30:
I meet the same problem. Have you solved it?

MusLead, 2023-04-30:
So, I hope they find solution for this. The biggest problem is because this RISC-V is not designed for ARM64 architecture. Therfore we need to do something.

The steps is also from ChatGPT generated.

I assume you have installed arch x86-64 and Rosetta x64 on you mac M1. The follow this instruction bellow:

This is for those who are trying to install RISC-V on Mac with ARM-Based Processor. You should be aware that RISC-V cannot be install just with *brew install riscv-tools* it will return some error, becuase this RISC-V is designed for x86_64 architecture. therefore,  there must be some some additional command.

'''
arch --x86_64 /usr/local/bin/brew tap riscv/riscv
arch --x86_64 /usr/local/bin/brew install riscv-gnu-toolchain
'''

This command actually works for me to install and to compile later on using *make qemu*.  I hope this solution can also help you guys, for those who are using Mac with processor M1/M2. If there are some suggestion for me, please let me know. Probably there is another alternative solution that better than mine.

gmerlino, 2023-04-30:
> I meet the same problem. Have you solved it?

I created the pull request above, because it fixes the issue (both the compilation error, and the resulting binary (pk) works fine) under ARM64 (Apple Silicon) by modifying the compiler options accordingly ("zifencei" explicitly enabled)

rain7996, 2023-04-30:
> > I meet the same problem. Have you solved it?
> 
> I created the pull request above, because it fixes the issue (both the compilation error, and the resulting binary - pk - works fine) under ARM64 (Apple Silicon) by modifying the compiler options accordingly ("zifencei" explicitly enabled)

> how can I use your version? Does the two command 'brew tape gmerlino/homebrew-riscv' and 'brew install riskv-tools' work fine?

sbeamer, 2023-05-25:
Can someone confirm this is still an issue? riscv-pk reverted the commit that caused this issue: https://github.com/riscv-software-src/riscv-pk/commit/acbe166dac9d1db752ee95f61e65ca82bb875afb

Justus2308, 2023-05-27:
It still doesn't work for me, I get the same error on a M1 MacBook Pro
```
...

../machine/flush_icache.c: Assembler messages:
../machine/flush_icache.c:4: Error: unrecognized opcode `fence.i', extension `zifencei' required
make: *** [flush_icache.o] Error 1
make: *** Waiting for unfinished jobs....
../machine/mentry.S: Assembler messages:
../machine/mentry.S:93: Error: unrecognized opcode `fence.i', extension `zifencei' required
make: *** [mentry.o] Error 1
```

rain7996, 2023-05-27:
gmerlino's solution works fine with me. I use the commands 'brew tape gmerlino/homebrew-riscv' and 'brew install riskv-tools' instead of the official.

derekcasamenti, 2023-06-05:
> gmerlino's solution works fine with me. I use the commands 'brew tape gmerlino/homebrew-riscv' and 'brew install riskv-tools' instead of the official. I thought this issue could be closed.

Same for me. I'm using a MacBook Pro 16 (M1 Pro) with macOS Ventura 13.4.

sbeamer, 2023-06-06:
After merging #101 should ok (for now)

closed 2023-06-06 by sbeamer