riscv-software-src/homebrew-riscv #102: Failed to build on Apple Silicon (M1)

howjmay, 2023-08-23:
The error message is listed as following
```
==> make
Last 15 lines from /Users/hauyang/Library/Logs/Homebrew/riscv-gnu-toolchain/07.make:
/usr/bin/install -c -m 644 ../.././gcc/gcc/varasm.h /opt/homebrew/Cellar/riscv-gnu-toolchain/main/lib/gcc/riscv64-unknown-elf/12.2.0/plugin/include/varasm.h
/usr/bin/install -c -m 644 ../.././gcc/gcc/vec-perm-indices.h /opt/homebrew/Cellar/riscv-gnu-toolchain/main/lib/gcc/riscv64-unknown-elf/12.2.0/plugin/include/vec-perm-indices.h
/usr/bin/install -c -m 644 ../.././gcc/gcc/vec.h /opt/homebrew/Cellar/riscv-gnu-toolchain/main/lib/gcc/riscv64-unknown-elf/12.2.0/plugin/include/vec.h
/usr/bin/install -c -m 644 ../.././gcc/gcc/vector-builder.h /opt/homebrew/Cellar/riscv-gnu-toolchain/main/lib/gcc/riscv64-unknown-elf/12.2.0/plugin/include/vector-builder.h
/usr/bin/install -c -m 644 version.h /opt/homebrew/Cellar/riscv-gnu-toolchain/main/lib/gcc/riscv64-unknown-elf/12.2.0/plugin/include/version.h
/usr/bin/install -c -m 644 ../.././gcc/gcc/vmsdbg.h /opt/homebrew/Cellar/riscv-gnu-toolchain/main/lib/gcc/riscv64-unknown-elf/12.2.0/plugin/include/vmsdbg.h
/usr/bin/install -c -m 644 ../.././gcc/gcc/vr-values.h /opt/homebrew/Cellar/riscv-gnu-toolchain/main/lib/gcc/riscv64-unknown-elf/12.2.0/plugin/include/vr-values.h
/usr/bin/install -c -m 644 ../.././gcc/gcc/vtable-verify.h /opt/homebrew/Cellar/riscv-gnu-toolchain/main/lib/gcc/riscv64-unknown-elf/12.2.0/plugin/include/vtable-verify.h
/usr/bin/install -c -m 644 ../.././gcc/gcc/wide-int-bitmask.h /opt/homebrew/Cellar/riscv-gnu-toolchain/main/lib/gcc/riscv64-unknown-elf/12.2.0/plugin/include/wide-int-bitmask.h
/usr/bin/install -c -m 644 ../.././gcc/gcc/wide-int-print.h /opt/homebrew/Cellar/riscv-gnu-toolchain/main/lib/gcc/riscv64-unknown-elf/12.2.0/plugin/include/wide-int-print.h
/usr/bin/install -c -m 644 ../.././gcc/gcc/wide-int.h /opt/homebrew/Cellar/riscv-gnu-toolchain/main/lib/gcc/riscv64-unknown-elf/12.2.0/plugin/include/wide-int.h
/usr/bin/install -c -m 644 ../.././gcc/gcc/xcoff.h /opt/homebrew/Cellar/riscv-gnu-toolchain/main/lib/gcc/riscv64-unknown-elf/12.2.0/plugin/include/xcoff.h
/usr/bin/install -c -m 644 ../.././gcc/gcc/xcoffout.h /opt/homebrew/Cellar/riscv-gnu-toolchain/main/lib/gcc/riscv64-unknown-elf/12.2.0/plugin/include/xcoffout.h
/usr/bin/install -c -m 644 b-header-vars /opt/homebrew/Cellar/riscv-gnu-toolchain/main/lib/gcc/riscv64-unknown-elf/12.2.0/plugin/include/b-header-vars
mkdir -p stamps/ && touch stamps/build-gcc-newlib-stage1

If reporting this issue please do so at (not Homebrew/brew or Homebrew/homebrew-core):
  https://github.com/riscv/homebrew-riscv/issues

/opt/homebrew/Library/Homebrew/utils/github/api.rb:337:in `raise_error': Validation Failed: [{"message"=>"The listed users and repositories cannot be searched either because the resources do not exist or you do not have permission to view them.", "resource"=>"Search", "field"=>"q", "code"=>"invalid"}] (GitHub::API::ValidationFailedError)
	from /opt/homebrew/Library/Homebrew/utils/github/api.rb:263:in `open_rest'
	from /opt/homebrew/Library/Homebrew/utils/github.rb:178:in `search'
	from /opt/homebrew/Library/Homebrew/utils/github.rb:182:in `search_results_items'
	from /opt/homebrew/Library/Homebrew/utils/github.rb:36:in `search_issues'
	from /opt/homebrew/Library/Homebrew/utils/github.rb:62:in `issues_for_formula'
	from /opt/homebrew/Library/Homebrew/exceptions.rb:502:in `fetch_issues'
	from /opt/homebrew/Library/Homebrew/exceptions.rb:497:in `issues'
	from /opt/homebrew/Library/Homebrew/exceptions.rb:553:in `dump'
	from /opt/homebrew/Library/Homebrew/brew.rb:151:in `rescue in <main>'
	from /opt/homebrew/Library/Homebrew/brew.rb:139:in `<main>'
/opt/homebrew/Library/Homebrew/formula.rb:2741:in `block in system': Failed executing: make (BuildError)
	from /opt/homebrew/Library/Homebrew/formula.rb:2677:in `open'
	from /opt/homebrew/Library/Homebrew/formula.rb:2677:in `system'
	from /opt/homebrew/Library/Taps/riscv/homebrew-riscv/riscv-gnu-toolchain.rb:71:in `install'
	from /opt/homebrew/Library/Homebrew/build.rb:177:in `block (3 levels) in install'
	from /opt/homebrew/Library/Homebrew/extend/kernel.rb:499:in `with_env'
	from /opt/homebrew/Library/Homebrew/build.rb:139:in `block (2 levels) in install'
	from /opt/homebrew/Library/Homebrew/formula.rb:1420:in `block in brew'
	from /opt/homebrew/Library/Homebrew/formula.rb:2908:in `block (2 levels) in stage'
	from /opt/homebrew/Library/Homebrew/extend/kernel.rb:499:in `with_env'
	from /opt/homebrew/Library/Homebrew/formula.rb:2907:in `block in stage'
	from /opt/homebrew/Library/Homebrew/resource.rb:128:in `block (2 levels) in unpack'
	from /opt/homebrew/Library/Homebrew/download_strategy.rb:113:in `chdir'
	from /opt/homebrew/Library/Homebrew/download_strategy.rb:105:in `stage'
	from /opt/homebrew/Library/Homebrew/resource.rb:124:in `block in unpack'
	from /opt/homebrew/Library/Homebrew/mktemp.rb:75:in `block in run'
	from /opt/homebrew/Library/Homebrew/mktemp.rb:75:in `chdir'
	from /opt/homebrew/Library/Homebrew/mktemp.rb:75:in `run'
	from /opt/homebrew/Library/Homebrew/resource.rb:228:in `stage_resource'
	from /opt/homebrew/Library/Homebrew/resource.rb:123:in `unpack'
	from /opt/homebrew/Library/Homebrew/resource.rb:97:in `stage'
	from /System/Library/Frameworks/Ruby.framework/Versions/2.6/usr/lib/ruby/2.6.0/forwardable.rb:230:in `stage'
	from /opt/homebrew/Library/Homebrew/formula.rb:2887:in `stage'
	from /opt/homebrew/Library/Homebrew/formula.rb:1413:in `brew'
	from /opt/homebrew/Library/Homebrew/build.rb:133:in `block in install'
	from /opt/homebrew/Library/Homebrew/extend/kernel.rb:499:in `with_env'
	from /opt/homebrew/Library/Homebrew/build.rb:128:in `install'
	from /opt/homebrew/Library/Homebrew/build.rb:229:in `<main>'
```

sbeamer, 2023-08-28:
Do any of the comments in #99 or #101 help?

sbeamer, 2023-09-25:
This appears to be using another repo (`gmerlino/riscv/riscv-gnu-toolchain`), so I would follow up with that repo.

closed 2023-09-25 by sbeamer

howjmay, 2023-10-04:
Sorry @sbeamer I think I didn't understand well.
I thought the solution in the issues you provided was using another fomula.

Here is the log I ran again
It is this repo, and the error message doesn't explain a lot. 
May I ask how to solve it?

```bash
$ brew install riscv-tools
==> Downloading https://formulae.brew.sh/api/cask.jws.json
######################################################################### 100.0%
==> Fetching dependencies for riscv-software-src/riscv/riscv-tools: riscv-software-src/riscv/riscv-gnu-toolchain, riscv-software-src/riscv/riscv-isa-sim and riscv-software-src/riscv/riscv-pk
==> Fetching riscv-software-src/riscv/riscv-gnu-toolchain
==> Cloning https://github.com/riscv/riscv-gnu-toolchain.git
Updating /Users/john/Library/Caches/Homebrew/riscv-gnu-toolchain--git
From github.com:riscv/riscv-gnu-toolchain
   5437780..12143b6  master     -> origin/master
==> Checking out branch master
Already on 'master'
Your branch is behind 'origin/master' by 9 commits, and can be fast-forwarded.
  (use "git pull" to update your local branch)
HEAD is now at 12143b6 Bump binutils to v2.41
==> Fetching riscv-software-src/riscv/riscv-isa-sim
==> Cloning https://github.com/riscv/riscv-isa-sim.git
Updating /Users/john/Library/Caches/Homebrew/riscv-isa-sim--git
==> Checking out branch master
Already on 'master'
Your branch is up to date with 'origin/master'.
HEAD is now at d1efcdff Merge pull request #1473 from riscv-software-src/unavailable
==> Fetching riscv-software-src/riscv/riscv-pk
==> Cloning https://github.com/riscv/riscv-pk.git
Cloning into '/Users/john/Library/Caches/Homebrew/riscv-pk--git'...
==> Checking out branch master
Already on 'master'
Your branch is up to date with 'origin/master'.
==> Fetching riscv-software-src/riscv/riscv-tools
==> Downloading https://github.com/riscv/riscv-tools/archive/homebrew.tar.gz
==> Downloading from https://codeload.github.com/riscv-software-src/riscv-tools/
 -#O#-  #   #
==> Installing riscv-tools from riscv-software-src/riscv
==> Installing dependencies for riscv-software-src/riscv/riscv-tools: riscv-software-src/riscv/riscv-gnu-toolchain, riscv-software-src/riscv/riscv-isa-sim and riscv-software-src/riscv/riscv-pk
==> Installing riscv-software-src/riscv/riscv-tools dependency: riscv-softw
==> git submodule update --init --recursive newlib
==> git submodule update --init --recursive binutils
^C
➜  Work brew install riscv-tools
==> Downloading https://formulae.brew.sh/api/formula.jws.json
######################################################################### 100.0%
==> Downloading https://formulae.brew.sh/api/cask.jws.json
######################################################################### 100.0%
==> Fetching dependencies for riscv-software-src/riscv/riscv-tools: riscv-software-src/riscv/riscv-gnu-toolchain, riscv-software-src/riscv/riscv-isa-sim and riscv-software-src/riscv/riscv-pk
==> Fetching riscv-software-src/riscv/riscv-gnu-toolchain
==> Cloning https://github.com/riscv/riscv-gnu-toolchain.git
Updating /Users/john/Library/Caches/Homebrew/riscv-gnu-toolchain--git
==> Checking out branch master
Already on 'master'
Your branch is up to date with 'origin/master'.
HEAD is now at 12143b6 Bump binutils to v2.41
==> Fetching riscv-software-src/riscv/riscv-isa-sim
==> Cloning https://github.com/riscv/riscv-isa-sim.git
Updating /Users/john/Library/Caches/Homebrew/riscv-isa-sim--git
==> Checking out branch master
Already on 'master'
Your branch is up to date with 'origin/master'.
HEAD is now at d1efcdff Merge pull request #1473 from riscv-software-src/unavailable
==> Fetching riscv-software-src/riscv/riscv-pk
==> Cloning https://github.com/riscv/riscv-pk.git
Updating /Users/john/Library/Caches/Homebrew/riscv-pk--git
==> Checking out branch master
Already on 'master'
[... 10 lines cut ...]
==> git submodule update --init --recursive gcc
==> sed -i .bak s/.*=host-darwin.o$// gcc/gcc/config.host
==> sed -i .bak s/.* x-darwin.$// gcc/gcc/config.host
==> ./configure --with-cmodel=medany --enable-multilib
==> make
Last 15 lines from /Users/john/Library/Logs/Homebrew/riscv-gnu-toolchain/07.make:
	    if [ -f /opt/homebrew/Cellar/riscv-gnu-toolchain/main/riscv64-unknown-elf/include/$i ]; then \
		rm /opt/homebrew/Cellar/riscv-gnu-toolchain/main/riscv64-unknown-elf/include/$i; \
	    else true; fi ; \
	  done ; \
	else true; fi
mkdir /private/tmp/riscv-gnu-toolchain-20231004-95864-fvjy9m/install-newlib-nano/riscv64-unknown-elf/include/rpc
mkdir /private/tmp/riscv-gnu-toolchain-20231004-95864-fvjy9m/install-newlib-nano/riscv64-unknown-elf/include/ssp
mkdir /private/tmp/riscv-gnu-toolchain-20231004-95864-fvjy9m/install-newlib-nano/riscv64-unknown-elf/include/sys
mkdir /opt/homebrew/Cellar/riscv-gnu-toolchain/main/riscv64-unknown-elf/include/rpc
mkdir /opt/homebrew/Cellar/riscv-gnu-toolchain/main/riscv64-unknown-elf/include/ssp
mkdir /opt/homebrew/Cellar/riscv-gnu-toolchain/main/riscv64-unknown-elf/include/sys
mkdir /private/tmp/riscv-gnu-toolchain-20231004-95864-fvjy9m/install-newlib-nano/riscv64-unknown-elf/include/bits
mkdir -p stamps/ && touch stamps/build-newlib-nano
mkdir /opt/homebrew/Cellar/riscv-gnu-toolchain/main/riscv64-unknown-elf/include/bits
mkdir -p stamps/ && touch stamps/build-newlib

If reporting this issue please do so at (not Homebrew/brew or Homebrew/homebrew-core):
  https://github.com/riscv-software-src/homebrew-riscv/issues
```

yulu83, 2023-11-02:
how can this issue be closed ?