riscv-software-src/homebrew-riscv #103: Failed to build on Intel MAC 12.6.8 Monterey

oxy1914, 2023-09-08:
user@MacBook-Pro-Aleksandr blink % brew install riscv-tools
==> Fetching dependencies for riscv-software-src/riscv/riscv-tools: riscv-software-src/riscv/riscv-gnu-toolchain, dtc, riscv-software-src/riscv/riscv-isa-sim and riscv-software-src/riscv/riscv-pk
==> Fetching riscv-software-src/riscv/riscv-gnu-toolchain
==> Cloning https://github.com/riscv/riscv-gnu-toolchain.git
Updating /Users/user/Library/Caches/Homebrew/riscv-gnu-toolchain--git
==> Checking out branch master
Your branch is up to date with 'origin/master'.
Already on 'master'
HEAD is now at da18f1c Merge pull request #1319 from cy023/patch-1
==> Fetching dtc
==> Downloading https://ghcr.io/v2/homebrew/core/dtc/manifests/1.7.0
Already downloaded: /Users/user/Library/Caches/Homebrew/downloads/00668efdbca5875f420621ff3011ae53448a6324636c2a27fe2323933c1ef3fa--dtc-1.7.0.bottle_manifest.json
==> Downloading https://ghcr.io/v2/homebrew/core/dtc/blobs/sha256:a319159e5002540c731072ab606432bfb116e2dc6606a3086be1b76c7bee0f68
Already downloaded: /Users/user/Library/Caches/Homebrew/downloads/8a90c31c6bcabb5d6f56dca483673e4bb6beb3f0d4a8a2cdf403613890cd55bd--dtc--1.7.0.monterey.bottle.tar.gz
==> Fetching riscv-software-src/riscv/riscv-isa-sim
==> Cloning https://github.com/riscv/riscv-isa-sim.git
Updating /Users/user/Library/Caches/Homebrew/riscv-isa-sim--git
==> Checking out branch master
Already on 'master'
Your branch is up to date with 'origin/master'.
HEAD is now at eb9a55a5 Merge pull request #1453 from riscv-software-src/attempt-to-fix-mac-ci
==> Fetching riscv-software-src/riscv/riscv-pk
==> Cloning https://github.com/riscv/riscv-pk.git
Updating /Users/user/Library/Caches/Homebrew/riscv-pk--git
==> Checking out branch master
Already on 'master'
Your branch is up to date with 'origin/master'.
HEAD is now at acbe166 Revert "[GCC]: Fix fence.i bug (#296)"
==> Fetching riscv-software-src/riscv/riscv-tools
==> Downloading https://github.com/riscv/riscv-tools/archive/homebrew.tar.gz
Already downloaded: /Users/user/Library/Caches/Homebrew/downloads/5e080ac67877162e34a9e6701d88df637ec2f4abb27be9857918a142c6a3477e--riscv-tools-homebrew.tar.gz
==> Installing riscv-tools from riscv-software-src/riscv
==> Installing dependencies for riscv-software-src/riscv/riscv-tools: riscv-software-src/riscv/riscv-gnu-toolchain, dtc, riscv-software-src/riscv/riscv-isa-sim and riscv-software-src/riscv/riscv-pk
==> Installing riscv-software-src/riscv/riscv-tools dependency: riscv-software-src/riscv/riscv-gnu-toolchain
==> git submodule update --init --recursive newlib
==> git submodule update --init --recursive binutils
==> git submodule update --init --recursive gcc
==> sed -i .bak s/.*=host-darwin.o$// gcc/gcc/config.host
==> sed -i .bak s/.* x-darwin.$// gcc/gcc/config.host
==> ./configure --with-cmodel=medany --enable-multilib
==> make
Last 15 lines from /Users/user/Library/Logs/Homebrew/riscv-gnu-toolchain/07.make:
	   riscv64-unknown-elf-ar x ../$i; \
	 done; \
	riscv64-unknown-elf-ar rc ../libc.a *.o
riscv64-unknown-elf-ar: ../stdio/lib.a: No such file or directory
riscv64-unknown-elf-ranlib libc.a
rm -rf tmp
make[8]: *** [all-recursive] Error 1
make[7]: *** [all-recursive] Error 1
make[6]: *** [all] Error 2
make[5]: *** [multi-do] Error 1
make[4]: *** [all-multi] Error 2
make[3]: *** [all] Error 2
make[2]: *** [all-target-newlib] Error 2
make[1]: *** [all] Error 2
make: *** [stamps/build-newlib-nano] Error 2

If reporting this issue please do so at (not Homebrew/brew or Homebrew/homebrew-core):
  https://github.com/riscv-software-src/homebrew-riscv/issues

sbeamer, 2023-09-25:
I would try debugging your system by installing the compiler directly (https://github.com/riscv-collab/riscv-gnu-toolchain/issues/1332) without homebrew. The snippet shared above cuts out the actual error.

sbeamer, 2024-01-11:
This might be the same issue as #116.

closed 2024-01-11 by sbeamer