riscv-software-src/homebrew-riscv #111: Install Failed for M1 Mac

WhoShotCupidd, 2023-10-18:
Steps:

brew tap riscv-software-src/riscv 
brew install riscv-tools 

Error:
==> Installing riscv-tools from riscv-software-src/riscv
==> Installing dependencies for riscv-software-src/riscv/riscv-tools: riscv-software-src/riscv/riscv-gnu-toolchain, riscv-software-src/riscv/riscv-isa-sim and riscv-software-src/riscv/riscv-pk
==> Installing riscv-software-src/riscv/riscv-tools dependency: riscv-software-src/riscv/riscv-gnu-toolchain
==> git submodule update --init --recursive newlib
==> git submodule update --init --recursive binutils
==> git submodule update --init --recursive gcc
==> sed -i .bak s/.*=host-darwin.o$// gcc/gcc/config.host
==> sed -i .bak s/.* x-darwin.$// gcc/gcc/config.host
==> ./configure --with-cmodel=medany --enable-multilib
==> make
Last 15 lines from /Users/christiaan/Library/Logs/Homebrew/riscv-gnu-toolchain/07.make:
	    if [ -f $i ]; then \
	     /usr/bin/install -c -m 644 $i /opt/homebrew/Cellar/riscv-gnu-toolchain/main/riscv64-unknown-elf/include/machine/`basename $i`; \
	    else true; fi ; \
	  done ; \
	  for i in rpc/types.h rpc/xdr.h; do \
	    if [ -f /opt/homebrew/Cellar/riscv-gnu-toolchain/main/riscv64-unknown-elf/include/$i ]; then \
		rm /opt/homebrew/Cellar/riscv-gnu-toolchain/main/riscv64-unknown-elf/include/$i; \
	    else true; fi ; \
	  done ; \
	else true; fi
mkdir /opt/homebrew/Cellar/riscv-gnu-toolchain/main/riscv64-unknown-elf/include/rpc
mkdir /opt/homebrew/Cellar/riscv-gnu-toolchain/main/riscv64-unknown-elf/include/ssp
mkdir /opt/homebrew/Cellar/riscv-gnu-toolchain/main/riscv64-unknown-elf/include/sys
mkdir /opt/homebrew/Cellar/riscv-gnu-toolchain/main/riscv64-unknown-elf/include/bits
mkdir -p stamps/ && touch stamps/build-newlib

If reporting this issue please do so at (not Homebrew/brew or Homebrew/homebrew-core):
  https://github.com/riscv-software-src/homebrew-riscv/issues

berserker1, 2023-10-18:
Same errror for M2 chip, what I do not understand is the error code is not exactly shown.

seanok, 2023-10-20:
Same error on MacBook Air M1

seanok, 2023-10-21:
I was able to install riscv-isa-sim without any errors. But I cannot get riscv-gnu-toolchain to build.  I tried with the no -32-bit target option and got the following output:

❯ brew install riscv-gnu-toolchain --with-NOmultilib
==> Downloading https://formulae.brew.sh/api/formula.jws.json
###################################################################################### 100.0%
==> Downloading https://formulae.brew.sh/api/cask.jws.json
#=#=-#
==> Fetching riscv-software-src/riscv/riscv-gnu-toolchain
==> Cloning https://github.com/riscv/riscv-gnu-toolchain.git
Updating /Users/sean/Library/Caches/Homebrew/riscv-gnu-toolchain--git
==> Checking out branch master
Already on 'master'
Your branch is up to date with 'origin/master'.
HEAD is now at b86b2b3 Switch from Ubuntu 20.04 LTS to 22.04 LTS for jobs (#1353)
==> Installing riscv-gnu-toolchain from riscv-software-src/riscv
==> git submodule update --init --recursive newlib
==> git submodule update --init --recursive binutils
==> git submodule update --init --recursive gcc
==> sed -i .bak s/.*=host-darwin.o$// gcc/gcc/config.host
==> sed -i .bak s/.* x-darwin.$// gcc/gcc/config.host
==> ./configure --with-cmodel=medany
==> make
Last 15 lines from /Users/sean/Library/Logs/Homebrew/riscv-gnu-toolchain/07.make:
/usr/bin/install -c -m 644 ../.././gcc/gcc/value-relation.h /opt/homebrew/Cellar/riscv-gnu-toolchain/main/lib/gcc/riscv64-unknown-elf/13.2.0/plugin/include/value-relation.h
/usr/bin/install -c -m 644 ../.././gcc/gcc/varasm.h /opt/homebrew/Cellar/riscv-gnu-toolchain/main/lib/gcc/riscv64-unknown-elf/13.2.0/plugin/include/varasm.h
/usr/bin/install -c -m 644 ../.././gcc/gcc/vec-perm-indices.h /opt/homebrew/Cellar/riscv-gnu-toolchain/main/lib/gcc/riscv64-unknown-elf/13.2.0/plugin/include/vec-perm-indices.h
/usr/bin/install -c -m 644 ../.././gcc/gcc/vec.h /opt/homebrew/Cellar/riscv-gnu-toolchain/main/lib/gcc/riscv64-unknown-elf/13.2.0/plugin/include/vec.h
/usr/bin/install -c -m 644 ../.././gcc/gcc/vector-builder.h /opt/homebrew/Cellar/riscv-gnu-toolchain/main/lib/gcc/riscv64-unknown-elf/13.2.0/plugin/include/vector-builder.h
/usr/bin/install -c -m 644 version.h /opt/homebrew/Cellar/riscv-gnu-toolchain/main/lib/gcc/riscv64-unknown-elf/13.2.0/plugin/include/version.h
/usr/bin/install -c -m 644 ../.././gcc/gcc/vmsdbg.h /opt/homebrew/Cellar/riscv-gnu-toolchain/main/lib/gcc/riscv64-unknown-elf/13.2.0/plugin/include/vmsdbg.h
/usr/bin/install -c -m 644 ../.././gcc/gcc/vr-values.h /opt/homebrew/Cellar/riscv-gnu-toolchain/main/lib/gcc/riscv64-unknown-elf/13.2.0/plugin/include/vr-values.h
/usr/bin/install -c -m 644 ../.././gcc/gcc/vtable-verify.h /opt/homebrew/Cellar/riscv-gnu-toolchain/main/lib/gcc/riscv64-unknown-elf/13.2.0/plugin/include/vtable-verify.h
/usr/bin/install -c -m 644 ../.././gcc/gcc/wide-int-bitmask.h /opt/homebrew/Cellar/riscv-gnu-toolchain/main/lib/gcc/riscv64-unknown-elf/13.2.0/plugin/include/wide-int-bitmask.h
/usr/bin/install -c -m 644 ../.././gcc/gcc/wide-int-print.h /opt/homebrew/Cellar/riscv-gnu-toolchain/main/lib/gcc/riscv64-unknown-elf/13.2.0/plugin/include/wide-int-print.h
/usr/bin/install -c -m 644 ../.././gcc/gcc/wide-int.h /opt/homebrew/Cellar/riscv-gnu-toolchain/main/lib/gcc/riscv64-unknown-elf/13.2.0/plugin/include/wide-int.h
/usr/bin/install -c -m 644 ../.././gcc/gcc/xcoff.h /opt/homebrew/Cellar/riscv-gnu-toolchain/main/lib/gcc/riscv64-unknown-elf/13.2.0/plugin/include/xcoff.h
/usr/bin/install -c -m 644 b-header-vars /opt/homebrew/Cellar/riscv-gnu-toolchain/main/lib/gcc/riscv64-unknown-elf/13.2.0/plugin/include/b-header-vars
mkdir -p stamps/ && touch stamps/build-gcc-newlib-stage2

If reporting this issue please do so at (not Homebrew/brew or Homebrew/homebrew-core):
  https://github.com/riscv-software-src/homebrew-riscv/issues

howjmay, 2023-10-22:
I think this may be the error
https://github.com/riscv-collab/riscv-gnu-toolchain/issues/1358

howjmay, 2023-10-23:
We may refer how xPack solve the problems

https://github.com/xpack-dev-tools/riscv-none-elf-gcc-xpack/tree/xpack

redpanda3, 2023-10-25:
Experience the same error...

howjmay, 2023-10-27:
@redpanda3 @berserker1 @seanok @WhoShotCupidd 

Could you help me verify my PR? It works on my M1 machine.

https://github.com/riscv-software-src/homebrew-riscv/pull/105

WhoShotCupidd, 2023-10-27:
I get this new error now when applying that change.

==> make
Warning: Not overwriting share/gdb
Error: An exception occurred within a child process:
  Errno::ENOENT: No such file or directory @ apply2files - /opt/homebrew/Cellar/riscv-gnu-toolchain/main/share/info/annotate.info

redpanda3, 2023-10-28:
Me too, same error

igarnier, 2023-11-05:
FWIW I managed to build by changing the `rm` to `rm_rf` when deleting the info files.

paopaopoo, 2023-11-09:
I use M1 Max MBP, macos sonoma 14.0, and I encountered exactly the same info when i installed riscv toolchain through homebrew. I still haven't found the solution yet. Then I try to git clone the rocket-tools (https://github.com/chipsalliance/rocket-tools.git), and follow the ReadME to build toolchain locally. Then it reports an issue like this:
```
ld: Undefined symbols:
  _iconv, referenced from:
      __nl_find_msg in libintl.a[12](dcigettext.o)
  _iconv_close, referenced from:
      __nl_free_domain_conv in libintl.a[7](loadmsgcat.o)
  _iconv_open, referenced from:
      __nl_init_domain_conv in libintl.a[7](loadmsgcat.o)
```
Then I figure out it's because I install both homebrew and macports, and the libs are conflict. So I completely uninstall macports and all its related installed ports and stuff, and rebuild the toolchain. If there're compiling errors because of dependency loss or other info, I just install the dependencies using homebrew or do some modification according to the info. Then the toolchain is successfully compiled without an error. Now I can use the toolchain to cross compile. 

Things you may need to do to build gdb:
1. brew install byacc bison (will report error when build gdb if not install these)
2. add path of mpc, gmp, mprf and "--disable-werror" option on the fsf-binutils-gdb line in the build.sh:
3. add "--with-arch=rv64gc_zifencei" option on the riscv-pk line in build.sh:
```
#! /bin/bash
#
# Script to build RISC-V ISA simulator, proxy kernel, and GNU toolchain.
# Tools will be installed to $RISCV.

. build.common

echo "Starting RISC-V Toolchain build process"

check_version() {
    $1 --version | awk "NR==1 {if (\$NF>$2) {exit 0} exit 1}" || (
        echo $3 requires at least version $2 of $1. Aborting.
        exit 1
    )
}

check_version automake 1.14 "OpenOCD build"
check_version autoconf 2.64 "OpenOCD build"
(cd riscv-openocd && ./bootstrap)
build_project riscv-openocd --prefix=$RISCV --enable-remote-bitbang --enable-jtag_vpi --disable-werror --disable-jlink
build_project riscv-isa-sim --prefix=$RISCV
build_project riscv-gnu-toolchain --prefix=$RISCV --disable-gdb
build_project fsf-binutils-gdb --prefix=$RISCV --target=riscv64-unknown-elf --with-gmp=/opt/homebrew --with-mpfr=/opt/homebrew --with-mpc=/opt/homebrew --disable-werror
CC= CXX= build_project riscv-pk --prefix=$RISCV --host=riscv64-unknown-elf  --with-arch=rv64gc_zifencei
build_project riscv-tests --prefix=$RISCV/riscv64-unknown-elf

echo -e "\\nRISC-V Toolchain installation completed!"
```
4. use gcc installed through homebrew instead of clang:
`CC=gcc-13 CXX=g++-13 ./build.sh`
5. install bison yacc
```
brew install bison byacc
brew unlink bison
export PATH="$(brew --prefix bison)/bin:$PATH"
```

Maybe you could try compile the toolchain using the repository (https://github.com/chipsalliance/rocket-tools.git) instead of using homebrew.
Just FYI.

gnsoares, 2023-11-09:
> FWIW I managed to build by changing the `rm` to `rm_rf` when deleting the info files.

Can confirm @howjmay 's fix with `rm_rf` worked for me in installing riscv-gnu-toolchain (Ventura Apple Silicon).

berserker1, 2023-11-10:
@paopaopoo I applied your method on my mac M2 chip and got the following error

```
rm: conftest.dSYM: is a directory
rm: conftest.dSYM: is a directory
rm: conftest.dSYM: is a directory
sed: 1: "/^sim_install_/{s/^\(si ...": extra characters at the end of p command
rm: conftest.dSYM: is a directory
rm: conftest.dSYM: is a directory
rm: conftest.dSYM: is a directory
configure: WARNING: linux/perf_event.h missing or too old; some features may be unavailable.
configure: WARNING: libipt is missing or unusable; some features may be unavailable.
rm: conftest.dSYM: is a directory
rm: conftest.dSYM: is a directory
configure: WARNING: libdebuginfod is missing or unusable; some features may be unavailable.
configure: WARNING: linux/perf_event.h missing or too old; some features may be unavailable.
configure: WARNING: libipt is missing or unusable; some features may be unavailable.
configure: WARNING: babeltrace is missing or unusable; GDB is unable to read CTF data.
conflicts: 60 shift/reduce, 69 reduce/reduce

```

GCC version is 13

I checked in other places and people mention how this arises due to version mismatch or some old bug of rocket-tools which got fixed. So I am not exactly sure how to go from here.

Thanks!

paopaopoo, 2023-11-10:
> @paopaopoo I applied your method on my mac M2 chip and got the following error
>
> ```
> rm: conftest.dSYM: is a directory
> rm: conftest.dSYM: is a directory
> rm: conftest.dSYM: is a directory
> sed: 1: [...]

These are not errors? I alse see a lot of these kind of stuff, but the build can continue. I only try to solve a problem if it causes the building progress to stop and exit.

berserker1, 2023-11-10:
@paopaopoo my bad i just realize I did not post the whole snippet, the remaining error

```
../../gdb/cli/cli-cmds.c: In function 'int run_under_shell(const char*, int)':
../../gdb/cli/cli-cmds.c:930:20: warning: 'pid_t vfork()' is deprecated: Use posix_spawn or fork [-Wdeprecated-declarations]
  930 |   if ((pid = vfork ()) == 0)
      |              ~~~~~~^~
In file included from ../gnulib/import/unistd.h:40,
                 from ../gnulib/import/stdlib.h:100,
                 from ../../gdb/../gdbsupport/common-defs.h:93,
                 from ../../gdb/defs.h:26,
                 from ../../gdb/cli/cli-cmds.c:20:
/Library/Developer/CommandLineTools/SDKs/MacOSX14.sdk/usr/include/unistd.h:606:10: note: declared here
  606 | pid_t    vfork(void) __WATCHOS_PROHIBITED __TVOS_PROHIBITED;
      |          ^~~~~
m2-exp.c:163:13: error: expected identifier before numeric constant
  163 | #define DOT 302
      |             ^~~
../../gdb/split-name.h:34:3: note: in expansion of macro 'DOT'
   34 |   DOT,
      |   ^~~
m2-exp.c:163:13: error: expected '}' before numeric constant
  163 | #define DOT 302
      |             ^~~
../../gdb/split-name.h:34:3: note: in expansion of macro 'DOT'
   34 |   DOT,
      |   ^~~
In file included from ../../gdb/symtab.h:39,
                 from ../../gdb/expression.h:24,
                 from ../../gdb/m2-exp.y:41:
../../gdb/split-name.h:28:1: note: to match this '{'
   28 | {
      | ^
m2-exp.c:163:13: error: expected unqualified-id before numeric constant
  163 | #define DOT 302
      |             ^~~
../../gdb/split-name.h:34:3: note: in expansion of macro 'DOT'
   34 |   DOT,
      |   ^~~
../../gdb/split-name.h:37:1: error: expected declaration before '}' token
   37 | };
      | ^
../../gdb/symtab.h: In member function 'std::vector<std::basic_string_view<char> > ada_lookup_name_info::split_name() const':
../../gdb/symtab.h:139:64: error: 'UNDERSCORE' is not a member of 'split_style'
  139 |     return ::split_name (m_encoded_name.c_str (), split_style::UNDERSCORE);
      |                                                                ^~~~~~~~~~
../../gdb/symtab.h: In member function 'std::vector<std::basic_string_view<char> > lookup_name_info::split_name(language) const':
m2-exp.c:163:13: error: expected unqualified-id before numeric constant
  163 | #define DOT 302
      |             ^~~
../../gdb/symtab.h:308:30: note: in expansion of macro 'DOT'
  308 |         style = split_style::DOT;
      |                              ^~~
make[2]: *** [m2-exp.o] Error 1
make[1]: *** [all-gdb] Error 2
make: *** [all] Error 2

```

paopaopoo, 2023-11-10:
I guess it's because you didn't use the new bison, do the steup before running build.sh:
[image]

berserker1, 2023-11-10:
Really thanks, it worked!

sbeamer, 2024-01-12:
This may be resolved for 2 reasons:

1. I disabled building gdb by default since few folks used it in the cross-compiled setup this repo provides. Without it, you don't need to worry about upgrading bison and yacc.
2. I pushed the first set of Apple Silicon bottles, so many folks can probably use those and not have to worry about building.

seanok, 2024-01-14:
> This may be resolved for 2 reasons:
>
> I disabled building gdb by default since few folks used it in the cross-compiled setup this repo provides. Without it, you don't need to worry about upgrading bison and yacc.
> I pushed the first set of Apple Silicon bottles, so many folks can probably use those and not have to worry about building.

Thanks.  This worked for me.

closed 2024-01-15 by sbeamer